Azure/azure-sphere-samples #8: Unable to build ExternalMcuUpdateNrf52

muesliman, 2018-11-16:
Schweregrad	Code	Beschreibung	Projekt	Datei	Zeile	Unterdrückungszustand
Fehler		 implicit declaration of function 'Storage_OpenFileInImagePackage' [-Werror=implicit-function-declaration]
     self->fd = Storage_OpenFileInImagePackage(path);
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1.exe: all warnings being treated as errors	ExternalMcuUpdateNrf52	E:\proj\Workspace_IoT\sphere\azure-sphere-samples\Samples\ExternalMcuUpdateNrf52\AzureSphereApp\ExternalMcuUpdateNrf52\file_view.c	39

richardtaylorrt, 2018-11-16:
Hi - have you updated to the [latest SDK](https://docs.microsoft.com/en-us/azure-sphere/resources/release-notes-1811) that we just released?

muesliman, 2018-11-16:
Works! That was the correct solution. Thank you!

closed 2018-11-16 by muesliman