Azure/azure-sphere-samples #29: Unclear LSM6DS3 Address in I2C sample

simon-jun, 2019-03-06:
Hi, 
I have a question about the I2C sample in "azure-sphere-samples/Samples/LSM6DS3_I2C/LSM6DS3_I2C/main.c" 
In line 43 the address is declared as 0x6A (01101010). I checked the Datasheet and it says the address would be 0xD4 (11010100). Seems like the address is shifted by 1 but I don't know why.
Would be nice if someone could explain this to me. 

`// DocID026899 Rev 10, S6.1.1, I2C operation`
`// SDO is tied to ground so the least significant bit of the address is zero.`
`static const uint8_t lsm6ds3Address = 0x6A;`

Thanks in advance.

richardtaylorrt, 2019-03-07:
Hi, no problem, happy to explain.  The datasheet says:

> The Slave ADdress (SAD) associated to the LSM6DS3 is 110101xb. The SDO/SA0 pin can
> be used to modify the less significant bit of the device address. If the SDO/SA0 pin is
> connected to the supply voltage, LSb is ‘1’ (address 1101011b); else if the SDO/SA0 pin is
> connected to ground, the LSb value is ‘0’ (address **1101010b**). This solution permits to
> connect and address two different inertial modules to the same I2C bus.

I2C uses seven-bit addresses.  “**1101010b**” is a seven-bit binary value (=0x6A).  The last ‘b’ character stands for “binary”.  It is not a placeholder for the final bit.

On the wire, an additional bit is used to indicate read or write, but that is not visible to the application.

simon-jun, 2019-03-07:
Ok, I got it. 
Thank you very much.

closed 2019-03-07 by simon-jun