Azure/azure-sphere-samples #31: show iot central config not recognizing real device

xtreampb, 2019-03-15:
So when attempting to follow the azure sphere azure iot sample, i get all the way to the part where you copy the results of the command "ShowIoTCentralConfig".  When I run that, it recognizes my iot central application 'aadev' but states i only have simulated devices.  I've added my azure sphere with the correct device id.  I have also uploaded and validate my tenant cert.

My device status in IoT Central is set to Registered.  It is not blocked.  I have attempted to approve it manually with no results.

Is there a step missing, or is this a bug in IoT Central/Azure Sphere/ShowIoTCentralConfig?

xtreampb, 2019-03-15:
after reverse engineering the the tool (ShowIoTCentralConfig).  I was able to extract this error message:

```
...Error: Time:Fri Mar 15 16:58:58 2019 File:C:\azure-iot-sdk-c\provisioning_client\src\prov_device_ll_client.c Func:_prov_transport_process_json_reply Line:424 Provisioning Failure: OperationId: 4.5859b9bd28fd70a9.ba790ba3-77c5-47e5-958f-06ccb28d7d34 - Date: 2019-03-15T20:58:55.6623173Z - Msg: Custom allocation failed with status code: 401

Error: Time:Fri Mar 15 16:58:58 2019 File:C:\azure-iot-sdk-c\provisioning_client\src\prov_transport_mqtt_common.c Func:_prov_transport_common_mqtt_dowork Line:944 Unable to process registration reply.

Error: Time:Fri Mar 15 16:58:58 2019 File:C:\azure-iot-sdk-c\provisioning_client\src\prov_device_ll_client.c Func:_on_transport_registration_data Line:510 Failure retrieving data from the provisioning service

Failure encountered on registration PROV_DEVICE_RESULT_DEV_AUTH_ERROR

error: registration failed!
```

The first time i encountered this error i downloaded the C sdk for IoT Central and installed to so that the paths were valid but i got the same error when i ran it a second time

mikehall-ms, 2019-03-18:
Can you run the ShowIoTCentralConfig application, and when prompted for Work/School account say 'n'. Follow the prompts to provide the Scope Id, Primary SaaS Key, and a Device ID. Does the application then provide the IoT Hub Endpoint URL?

xtreampb, 2019-03-18:
it is not able to retrieve the hub endpoint information.  I think there is an issue with the embedded `dps_cstr` exe resource.  I got the above error from that tool.

mikehall-ms, 2019-03-18:
I'd like to get more information on the problem you are seeing, can you send me an e-mail at 'mikehall@microsoft.com' so we can discuss.

xtreampb, 2019-03-18:
Closed due to user error.

closed 2019-03-18 by xtreampb