Azure/azure-sphere-samples #32: AzueIoT Sample MSB3073 Error

piercalderan, 2019-03-24:
This is the output when compiling:
Gravità	Codice	Descrizione	Progetto	File	Riga	Stato eliminazione
Errore	MSB3073	uscita dal comando "%SYSTEMROOT%\System32\WindowsPowerShell\v1.0\powershell.exe -ExecutionPolicy Bypass -NoProfile -NonInteractive -File "C:\Users\pier\Desktop\azure-sphere-samples-master\Samples\AzureIoT\AzureIoT\script\validate_manifest.ps1
:VCEnd" con codice 1.	AzureIoT	C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	128

richardtaylorrt, 2019-03-25:
Hi - you need to configure the app to [work with your IoT Central](https://github.com/Azure/azure-sphere-samples/tree/master/Samples/AzureIoT#configure-the-sample-application-to-work-with-your-azure-sphere-tenant-and-devices) or to [work with your IoT Hub](https://github.com/Azure/azure-sphere-samples/tree/master/Samples/AzureIoT#configure-the-sample-application-to-work-with-your-azure-iot-hub).

If you switch to the "Output -> Build" panel in Visual Studio, you will see this advice:

```
1>------ Build started: Project: AzureIoT, Configuration: Debug ARM ------
1>Validating project app_manifest.json
1>CmdArgs is empty, this needs to contain the ScopeId for your IoT Hub/Central application
1>The app_manifest.json 'AllowedConnections' needs to contain:
1>  Your IoT Hub/Central application Endpoint address
1>The app_manifest.json file contains a placeholder 'DeviceAuthentication'
1>this needs to be replaced with your Azure Sphere Tenant Id
1>This can be obtained using the Azure Sphere Developer Command Prompt
1>'azsphere tenant show-selected'
...
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

```

Please let us know if this solves the problem, thanks.

piercalderan, 2019-03-25:
Hi,

Thanks for the info... now the ScopeID is OK.

But the output is:
IoT Hub/Central Application starting.
Setting Azure Scope ID 0ne0004F70C
Opening MT3620_RDB_BUTTON_A as input
Opening MT3620_RDB_BUTTON_B as input
Opening MT3620_RDB_LED4_BLUE as output
Sending IoT Hub Message: { "ButtonPress": "True" }
WARNING: failed to hand over the message to IoTHubClient

I tried to run: azsphere device show-attached

But the Device ID is this and it's unusable:
Device ID: 
CC882AAF51E5C47BC40ED1230AB5F647188DB2CC81AA91146487B7CC7A2E538A3B33D0D9AE0EE852F19076044E653F7D6DC6AC64D8618BC5D479156FF4C80BD1
Command completed successfully in 00:00:01.8087000.

and then I ran...ShowIotCentralConfig.
It shows this output, even if I created more than one REAL devices!
Getting your IoT Central applications
You have one IoT Central application 'App Pier'.
Getting the Device Provisioning Service (DPS) information.
Getting a list of IoT Central devices.
Please create a real device in your IoT Central application and try again.

Any idea?

-- 
Pier Calderan
*Electronics & Inventions:
* *Pierduino* Piccoli grandi progetti fai da te www.pierduino.com
*Cremona Musica (Artistic Advisor)* www.cremonamusica.com
*Audio Video & Music* www.audiovideomusic.com
*Tel.* +39 328 2117593
*Skype ID* pierduino <skype:pierduino>

***********************************************************************************************************************************
Le informazioni trasmesse attraverso la presente e-mail ed i suoi 
allegati sono diretti esclusivamente
al destinatario e devono ritenersi riservati con divieto di diffusione e 
di uso nei giudizi salva espressa autorizzazione
Nel caso di utilizzo senza espressa autorizzazione, verrà effettuata 
denuncia al competente Consiglio dell’Ordine
per violazione dell’art. 28 del Codice Deontologico. La diffusione e la 
comunicazione da parte di soggetto diverso
dal destinatario è vietata dall’art. 616 e ss. c.p. e dal d. l.vo n. 196/03.
Se la presente e-mail ed si suoi allegati fossero stati ricevuti per 
errore da persona diversa dal destinatario
siete pregati di distruggere tutto quanto ricevuto e di informare il 
mittente con lo stesso mezzo.
*********************************************************************************************************************************** 

---
Questa e-mail è stata controllata per individuare virus con Avast antivirus.
https://www.avast.com/antivirus

richardtaylorrt, 2019-03-25:
Hi - are you having problems at steps 11 and 12 of the [IoT Central configuration](https://github.com/Azure/azure-sphere-samples/tree/master/Samples/AzureIoT#run-the-sample-with-azure-iot-central)?  Note the need to run a powershell command to convert the device ID into lower case letters.  If this doesn't work, please send a screenshot of the error you see, or of your successfully created 'real' device in IoT Central.  Thanks.

piercalderan, 2019-03-25:
... even if I make lower-case the number id is not valid.

it should be in the form of:

ComponentId": "14d320f2-fd25-447b-858f-8d174cb5e931"

as required in app_manifest.json.

-- 
Pier Calderan
*Electronics & Inventions:
* *Pierduino* Piccoli grandi progetti fai da te www.pierduino.com
*Cremona Musica (Artistic Advisor)* www.cremonamusica.com
*Audio Video & Music* www.audiovideomusic.com
*Tel.* +39 328 2117593
*Skype ID* pierduino <skype:pierduino>

***********************************************************************************************************************************
Le informazioni trasmesse attraverso la presente e-mail ed i suoi 
allegati sono diretti esclusivamente
al destinatario e devono ritenersi riservati con divieto di diffusione e 
di uso nei giudizi salva espressa autorizzazione
Nel caso di utilizzo senza espressa autorizzazione, verrà effettuata 
denuncia al competente Consiglio dell’Ordine
per violazione dell’art. 28 del Codice Deontologico. La diffusione e la 
comunicazione da parte di soggetto diverso
dal destinatario è vietata dall’art. 616 e ss. c.p. e dal d. l.vo n. 196/03.
Se la presente e-mail ed si suoi allegati fossero stati ricevuti per 
errore da persona diversa dal destinatario
siete pregati di distruggere tutto quanto ricevuto e di informare il 
mittente con lo stesso mezzo.
*********************************************************************************************************************************** 

---
Questa e-mail è stata controllata per individuare virus con Avast antivirus.
https://www.avast.com/antivirus

piercalderan, 2019-03-25:
... the problem is here

azsphere device show-attached

dos not show device id

-- 
Pier Calderan
*Electronics & Inventions:
* *Pierduino* Piccoli grandi progetti fai da te www.pierduino.com
*Cremona Musica (Artistic Advisor)* www.cremonamusica.com
*Audio Video & Music* www.audiovideomusic.com
*Tel.* +39 328 2117593
*Skype ID* pierduino <skype:pierduino>

***********************************************************************************************************************************
Le informazioni trasmesse attraverso la presente e-mail ed i suoi 
allegati sono diretti esclusivamente
al destinatario e devono ritenersi riservati con divieto di diffusione e 
di uso nei giudizi salva espressa autorizzazione
Nel caso di utilizzo senza espressa autorizzazione, verrà effettuata 
denuncia al competente Consiglio dell’Ordine
per violazione dell’art. 28 del Codice Deontologico. La diffusione e la 
comunicazione da parte di soggetto diverso
dal destinatario è vietata dall’art. 616 e ss. c.p. e dal d. l.vo n. 196/03.
Se la presente e-mail ed si suoi allegati fossero stati ricevuti per 
errore da persona diversa dal destinatario
siete pregati di distruggere tutto quanto ricevuto e di informare il 
mittente con lo stesso mezzo.
*********************************************************************************************************************************** 

---
Questa e-mail è stata controllata per individuare virus con Avast antivirus.
https://www.avast.com/antivirus

richardtaylorrt, 2019-03-25:
The component ID is not the same as a device ID.
- You do _not_ need to change the component ID in the app manifest.  The existing value of "819255ff-8640-41fd-aea7-f85d34c491d5" will work fine.
- You _do_ need to use your device ID (128 hex digits in lower case) to create a real device in Azure IoT Central (steps 11 and 12).  Does this work for you?
- Once you have a real device, you can run "ShowIoTCentralConfig".  It will tell you what you need to put in your manifest.

If this still doesn't work, to help us to help you further, please could you attach:
1. a screenshot of your device created in IoT Central
2. the output from running "ShowIoTCentralConfig"
3. the output from the "Output -> Build" panel in Visual Studio

Thanks.

piercalderan, 2019-03-26:
Hi, here the result following your instructions (Please see the screenshots).
If your MT3260 is really working, please make the screenshot that you make on your side.
Thank you.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

richardtaylorrt, 2019-03-27:
Thanks for the additional information. There may be an issue with creating devices in IoT Central, which is impacting you. We are investigating and will update here as soon as we can.

m-chau, 2019-03-28:
I am having the same issue where I have created "real" devices in my IoT Central application but the ShowIoTCentralConfig.exe fails to find them. Please keep us updated on the resolution. Thank you.

richardtaylorrt, 2019-05-30:
Hi @piercalderan , @m-chau - some fixes have been made to IoT Central and to the "ShowIoTCentralConfig" tool.  Please can you pull the latest, retry, and let us know if any further problems in getting this app to run.  Thanks.

closed 2019-06-05 by cawhitworth