Azure/azure-sphere-samples #33: A compile error form example AzureIoT

david-foo, 2019-03-28:
Severity	Code	Description	Project	File	Line	Suppression State
Error	MSB3073	The command "%SYSTEMROOT%\System32\WindowsPowerShell\v1.0\powershell.exe -ExecutionPolicy Bypass -NoProfile -NonInteractive -File "D:\azure-sphere-samples\Samples\AzureIoT\AzureIoT\script\validate_manifest.ps1
:VCEnd" exited with code 1.	AzureIoT	C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	133

richardtaylorrt, 2019-03-28:
Hi - you need to configure the app to [work with your IoT Central](https://github.com/Azure/azure-sphere-samples/tree/master/Samples/AzureIoT#configure-the-sample-application-to-work-with-your-azure-sphere-tenant-and-devices) or to [work with your IoT Hub](https://github.com/Azure/azure-sphere-samples/tree/master/Samples/AzureIoT#configure-the-sample-application-to-work-with-your-azure-iot-hub).

If you switch to the "Output -> Build" panel in Visual Studio, you will see this advice:

```
1>------ Build started: Project: AzureIoT, Configuration: Debug ARM ------
1>Validating project app_manifest.json
1>CmdArgs is empty, this needs to contain the ScopeId for your IoT Hub/Central application
1>The app_manifest.json 'AllowedConnections' needs to contain:
1>  Your IoT Hub/Central application Endpoint address
1>The app_manifest.json file contains a placeholder 'DeviceAuthentication'
1>this needs to be replaced with your Azure Sphere Tenant Id
1>This can be obtained using the Azure Sphere Developer Command Prompt
1>'azsphere tenant show-selected'
...
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

```

Please let us know if this solves the problem, thanks.

david-foo, 2019-03-29:
@richardtaylorrt  
Thank you. The content in app_manifest.json can not be left blank when build the code.  Now the demo code can runs with IoT central.

closed 2019-03-29 by david-foo