Azure/azure-sphere-samples #45: CurlEasyHttps stuck on "setting TCP_NODELAY"

snobu, 2019-04-22:
Switched the CA bundle to the Mozilla one,

Hangs somewhere around here -

```
 // Perform the download of the web page.
    if ((res = curl_easy_perform(curlHandle)) != CURLE_OK) {
        LogCurlError("curl_easy_perform", res);
    } else {
        Log_Debug("\n -===- Downloaded content (%zu bytes): -===-\n", block.size);
        Log_Debug("%s\n", block.data);
    }
```
Target URL https://api.github.com/, same result with any *.azurewebsites.net site (try https://amisecure.azurewebsites.net).

Debug window output (https://api.github.com/):
```
Remote debugging from host 192.168.35.1
cURL easy interface based application starting.
This sample periodically attempts to download a webpage, using curl's 'easy' API.
 -===- Starting download -===-
*   Trying 192.30.253.117...
* TCP_NODELAY set
```

Tried to force TLS 1.2, same result, no go.
```
	if ((res = curl_easy_setopt(curlHandle, CURLOPT_SSLVERSION, CURL_SSLVERSION_TLSv1_2)) != CURLE_OK) {
		LogCurlError("curl_easy_setopt CURLOPT_SSLVERSION", res);
		goto cleanupLabel;
	}
```
https://example.com/ works fine and libcurl handshakes TLS 1.2 with remote.

richardtaylorrt, 2019-04-25:
Hi - apologies for slow response.  Did you add your target URLs to the "AllowedConnections" in the app_manifest.json?  This is explained [here](https://docs.microsoft.com/en-us/azure-sphere/app-development/curl#application-manifest).  (We will also add mention of this point to this sample app's README.)

snobu, 2019-04-25:
That's news to me. No other MCU/embedded Linux board has this clean network filtering capability in the SDK (which is probably one of the big selling points here). That said, you should **absolutely** add that piece of information to the sample's README and as code comments. We're engineers, we don't start with documentation :)

Everything is fine after amending `app_manifest.json`.

snobu, 2019-04-25:
One related question,

Is this CA list hand picked or same as [Mozilla's CA bundle](https://curl.haxx.se/docs/caextract.html)?
https://docs.microsoft.com/en-us/azure-sphere/app-development/curl#ca-certificates

richardtaylorrt, 2019-04-25:
Glad to hear it's working now.  I'll track the good idea of adding this to code comments too.

Unfortunately, there's no developer-mode way to switch off this filtering.

You can create your own TLS connection but, today, you'd need to bring your own TLS stack.  We're tracking a future feature request that would allow you to use our on-board TLS stack.  What's your scenario?  Please also suggest this feature on our [User Voice](https://feedback.azure.com/forums/915433-azure-sphere).

The [CA list is hand-picked](https://docs.microsoft.com/en-us/azure-sphere/app-development/curl#ca-certificates).  I see the Mozilla bundle is 250KB, which might be too big for us to absorb by default on our resource-constrained devices.

closed 2019-04-27 by snobu