Azure/azure-sphere-samples #47: where to find the Edit Template for adding new Measurement

BillyHsu, 2019-05-20:
Dear Sir,

I follow instruction from the readme of AzureIoT example.
I can't find the edit template from Device Explorer->select your device->device page in "Add new measurements, settings, and properties" segment.
Can you help to check the flow?
Thanks.

Best Regards,
Billy

richardtaylorrt, 2019-05-23:
Hi - the Azure IoT Central UI has recently changed.  There is now a specific entry for "Device Templates" on the left-hand-side of the UI.  You assign devices to a template using "Device Sets" that are also on the left-hand-side of the UI.  We will update the Azure Sphere docs to reflect this soon.

Please let us know if this doesn't give you enough information to adapt the instructions .

richardtaylorrt, 2019-08-09:
We're closing this issue now.  Please re-open if you still need assistance, and we'll be happy to help.  Thanks.

closed 2019-08-09 by richardtaylorrt