Azure/azure-sphere-samples #49: Linker scripts should ensure required vector table alignment

henrygab, 2019-06-04:
The ARM docs require the vector table be aligned on (at least) a 128-byte boundary.
<details>
  <summary>Show example source code comment</summary>

  <p>

https://github.com/Azure/azure-sphere-samples/blob/d232f0e8ab02dea57b2f2f96c13a4d5a2387acc0/Samples/ADC/ADC_RTApp_MT3620_BareMetal/main.c#L20-L23
</details>

However, it appears that the `linker.ld` files in the repo only **_guarantee_** 16-byte alignment.

<details>
  <summary>Click to show the five currently linker.ld files</summary>

  <p>

https://github.com/Azure/azure-sphere-samples/blob/d232f0e8ab02dea57b2f2f96c13a4d5a2387acc0/Samples/UART/UART_RTApp_MT3620_BareMetal/linker.ld#L18-L21

https://github.com/Azure/azure-sphere-samples/blob/d232f0e8ab02dea57b2f2f96c13a4d5a2387acc0/Samples/IntercoreComms/IntercoreComms_RTApp_MT3620_BareMetal/linker.ld#L18-L21

https://github.com/Azure/azure-sphere-samples/blob/d232f0e8ab02dea57b2f2f96c13a4d5a2387acc0/Samples/HelloWorld/HelloWorld_RTApp_MT3620_BareMetal/linker.ld#L18-L21

https://github.com/Azure/azure-sphere-samples/blob/d232f0e8ab02dea57b2f2f96c13a4d5a2387acc0/Samples/GPIO/GPIO_RTApp_MT3620_BareMetal/linker.ld#L18-L21

https://github.com/Azure/azure-sphere-samples/blob/d232f0e8ab02dea57b2f2f96c13a4d5a2387acc0/Samples/ADC/ADC_RTApp_MT3620_BareMetal/linker.ld#L18-L21

</details>

NOTE: Yes, I realize the files "work" at the moment.  However, it makes sense to avoid fragility in linker scripts, which includes explicitly listing the alignment requirements where needed, rather than relying on items being the first listed in a section, and the section being the first section, and the memory region itself thus happening to meet the alignment requirement....

I think either of the following would be a correct fix:
1. Create a new section, prior to listed one, with alignment specified as 128 instead of 16, and move `.vector_table` to that new section.
2. Just modify the alignment of the existing .text section to 128.  This has the potential to waste some bytes if that alignment is not needed for other entries of that section, but would still be correct.

Please confirm if this is a bug, and I'll gladly submit a PR.  I'd lean towards option #1 for the PR, so alternatives or arguments for/against either of the above (or a better solution altogether) are welcome!

henrygab, 2019-06-05:
After additional thoughts, neither of the above two are likely advisable.  Here's my current understanding:

* Adding a new section might break later, as some output formats restrict the allowable section names to a small known list. [Ref](https://sourceware.org/binutils/docs/ld/Output-Section-Name.html)
* Modifying the alignment of the section as a whole will cause only the first entry in that section to be aligned.  This is OK, but on the off chance that more than one symbol was defined to use` .vector_table`, the alignment would not be guaranteed for any but the first symbol.
* Use of SUBALIGN(128) would cause all entries in that section to be aligned at 128 byte boundaries.  This would ensure the interrupt table is properly aligned, but also waste significant memory because all symbols would be aligned at 128 byte boundaries.

What seems to be a correct fix is actually defining the alignment requirement on the interrupt vector table data structure itself.
<details>
  <summary>Currently, ExceptionVectorTable is defined as an array of uintptr_t</summary>

  <p>

https://github.com/Azure/azure-sphere-samples/blob/d232f0e8ab02dea57b2f2f96c13a4d5a2387acc0/Samples/ADC/ADC_RTApp_MT3620_BareMetal/main.c#L30-L45

</details>

It seems that it may be more appropriate to define a structure which is itself 128-byte aligned, which would then ensure it's placed in an appropriately-aligned memory location.
<details>
  <summary>Strawman / draft changes embodying the above concept</summary>

<p>

```c
typedef struct { uintptr_t v[EXCEPTION_COUNT]; } 
    __attribute__((section(".vector_table")))
    __attribute__((used))
    __attribute__((aligned(128)))
    EXCEPTION_VECTOR_TABLE;

static const EXCEPTION_VECTOR_TABLE 
    __attribute__((section(".vector_table")))
    __attribute__((used))
    __attribute__((aligned(128)))
    = {
        .v[0] = (uintptr_t)&StackTop, // Main Stack Pointer (MSP)
        .v[1] = (uintptr_t)RTCoreMain,               // Reset
        .v[2] = (uintptr_t)DefaultExceptionHandler,  // NMI
        .v[3] = (uintptr_t)DefaultExceptionHandler,  // HardFault
        .v[4] = (uintptr_t)DefaultExceptionHandler,  // MPU Fault
        .v[5] = (uintptr_t)DefaultExceptionHandler,  // Bus Fault
        .v[6] = (uintptr_t)DefaultExceptionHandler,  // Usage Fault
        .v[11] = (uintptr_t)DefaultExceptionHandler, // SVCall
        .v[12] = (uintptr_t)DefaultExceptionHandler, // Debug monitor
        .v[14] = (uintptr_t)DefaultExceptionHandler, // PendSV
        .v[15] = (uintptr_t)DefaultExceptionHandler, // SysTick
        .v[INT_TO_EXC(0)... INT_TO_EXC(INTERRUPT_COUNT - 1)] = (uintptr_t)DefaultExceptionHandler};
        };
```

</details>

I am not a linker guru, and haven't even tried to compile the above... rather the code's purpose is for discussion of ensuring the vector table gets the proper alignment when it's defined....

henrygab, 2019-06-06:
Here's one solution.  Comments or feedback?  I would make substantially the same changes to the other samples, and submit a PR.
https://github.com/henrygab/azure-sphere-
samples/commit/f9e1161a375eeb0d4e186fb898b031fff78e522d

Specifically, is there an easier way to (via preprocessor) generate the result currently provided by the CEIL_POW2() macro I added?

richardtaylorrt, 2019-06-06:
Thanks for looking in to this in such detail.  Our team is busy investigating, and we will respond here as soon as we can.

richardtaylorrt, 2019-06-10:
Our engineering team have looked into this. Here's what they say:

> The ARM documentation (ARM DDI0403E.d §B1.5.2-3) states the vector table must be aligned on a power-of-two boundary which is at least 128 bytes and no less than the size of the exception table.  The MT3620 supports 100 external interrupts so, including the stack pointer and 15 standard exceptions, it has 116 entries.  This gives a total size of 4 * 116 = 464 bytes, which rounds up to 512 bytes.
> 
> When the sample application is built for XIP flash, it must be placed at 0x10000000.  When it is built for TCM, it is typically placed at the start of TCM0, which is 0x100000.  In either case, the alignment is achieved by putting the vector table in a dedicated section, and setting CODE_REGION to the appropriate address.
> 
> The ALIGN attribute in the linker script is used to adjust where the vector table is physically located in the ELF file.  If the application is run from XIP flash, then it must be at an offset which is a multiple of 32 bytes.
> 

We will also update the M4 samples in this repo to explicitly use 32-byte alignment.

Please let us know if this answers your question adequately, thanks.

henrygab, 2019-06-11:
I didn't really have questions on the information just provided ... see my CEIL_POW2() macro, which ensured alignment at 512 bytes, even for XIP.  The mandatory address for XIP is admittedly somewhat of a surprise, simply because I hadn't found documentation of that requirement.

FYI, the vector table is not currently in a dedicated `section`, but rather simply the first listed item in the `.text` section.  Perhaps you are indicating the next update will provide a separate section for the exception vector table?

Based on your response, it appears that there is a high comfort level with the location & alignment being enforced via the linker file.  In this case, updating the M4 samples to explicitly use 32-byte alignment seems a "good enough" fix.  Shall I generate a PR?

cawhitworth, 2019-06-24:
I believe #52 addresses the issues raised here.

pennyo, 2019-06-25:
FWIW, the online docs have also been updated to reflect the alignment requirements:
https://docs.microsoft.com/azure-sphere/app-development/memory-latency#vector-table-location

henrygab, 2019-06-26:
#### Documentation changes
The linker script now has comments warning about the need for special alignment for the exception vectors.  The online documentation update also gives good details on this.
Thus, the documentation is now significantly better, and (imho) best-in-class on this arcane topic. Thanks!

#### Linker script / Alignment Enforcement
I'm still not sure that the changes explicitly enforce alignment of the vector table.  I've broken the comments into three parts: (1) correcting the section alignment requirement, (2) explicitly marking vector table alignment requirement, and (3) consideration of declaration of properly aligned struct.  Details follow for each, in order...

<details>
  <summary>Section alignment correction</summary>

<p>

The following line set the **_start of the section_** to be aligned to **32 bytes**:
https://github.com/Azure/azure-sphere-samples/blob/a024a1e9d07281d7fa9aae5714c9e98b844dfd3b/Samples/ADC/ADC_RTApp_MT3620_BareMetal/linker.ld#L31

By itself, this does not enforce any alignment requirement for any of the symbols within the section.  Since the vector table alignment must be **512 bytes**, this appears to rely on the predefined region (`CODE_REGION`) being aligned sufficiently.  (This happens to be true for this chip.)

However, if wanting to rely on the vector table being the first symbol in the section, would it be clearer to declare this section as follows?

`.text : ALIGN(512) {`

</details>
<details>
  <summary>Alignment directly at vector table</summary>

  <p>

The second line is the vector table, but it's written as simply being an array of pointers, and thus the structure itself has 4-byte alignment, not 512 byte alignment.  The following line does not ensure the structure is aligned properly, nor ensure that it's loaded to the required address:
https://github.com/Azure/azure-sphere-samples/blob/a024a1e9d07281d7fa9aae5714c9e98b844dfd3b/Samples/ADC/ADC_RTApp_MT3620_BareMetal/linker.ld#L31-L32

Would it make sense to be explicit about the alignment and loading address applying to the vector table?
```
    .text : ALIGN(32) {
        KEEP(*(.vector_table)) = ALIGN(512);
```
</details>

<details>
  <summary>Explicitly aligned struct?</summary>

  <p>

None of the above actually force the linker to ensure all symbols that match are aligned at the required boundary.  Is the linker actually the right place to enforce the alignment requirement?

I would suggest that, if a data structure must meet a given alignment requirement, then that alignment requirement should be explicit in the source code.   Currently, the vector table is defined as a simple array of `uint32_t`, and thus has only **_4-byte alignment_** required.

Here's an over-engineered example that dynamically determines the necessary alignment **_at preprocessor time_** (thus safe to use in static assert macro):
https://github.com/henrygab/azure-sphere-samples/commit/f9e1161a375eeb0d4e186fb898b031fff78e522d

An alternative is to declare the alignment to be 512, and static_assert that this is smaller than the structure size:
```C
typedef struct {
    uintptr_t v[EXCEPTION_COUNT];
}  __attribute__((aligned(512))) EXCEPTION_VECTOR_TABLE;
C_ASSERT(512 >= sizeof(EXCEPTION_VECTOR_TABLE));
```
</details>

#### Summary
1. Is it worth being more explicit on the alignment requirements?  Maybe.
2. Does the current linker script ensure the alignment requirement? Yes, because start of TCM is aligned, and table is first in section.
3. Does the current linker script **_explicitly_** ensure the alignment requirement? No ... in that it is only _implied_ by memory region address....

richardtaylorrt, 2019-07-02:
Hi - I talked further to our engineering team about this:

- We understand the layout is only true for the MT3620 chip.  Indeed, the entire sample is MT3620 specific and named accordingly!  If it were ported to another future Azure Sphere chip, many things would change.  In particular, it would require a new vector table because that chip would support different interrupts.  The new vector table would also have its own alignment requirements, depending on how many interrupts were supported on the new chip.  The sample would also require a new linker script, because the new chip would have different types of memory at different locations.

- We observer that the Zephyr RTOS, which aims to be more generic than our samples, also does [not specify](https://github.com/zephyrproject-rtos/zephyr/blob/master/arch/arm/core/cortex_m/vector_table.S) vector table alignment in the source code, but also [uses](https://github.com/zephyrproject-rtos/zephyr/blob/master/include/arch/arm/cortex_m/scripts/linker.ld#L166) a dedicated section.

- We could add explicit alignment via `__attribute__((aligned(512)))` but this would add 416 extra bytes to the imagepackage file without helping with the broader portability considerations above. 

Please let us know if you feel we're making the wrong trade-offs here between correctness, conciseness, explicit verboseness, and forwards-compatibility.  Thanks.

henrygab, 2019-07-13:
Hi @richardtaylorrt ,

We agree that the in-code __attribute__((aligned(512))) is not a good solution, for the precise reason you indicated (wasted space due to C causing struct size to increase to multiple of alignment).

I also think we can agree that low-level memory packing / location specification is … an area with growth potential for the embedded space.

What would I recommend, looking forward, is for each platform/board/chip:
1. explicit list of required dedicated sections and required address(es)
2. post-linkage validation, as part of build process, that each dedicated section...
   A. exists
   B. has exactly one symbol
   C. is stored at the required address (if applicable)
   D. loads at the required address (if applicable)

<details>
  <summary>Why the post-linkage validation?</summary>

<p>

We agree that the use of a **_dedicated_** section is good, especially if that section has the appropriate alignment specified.  

Unfortunately, there does not appear to be any current method to indicate a section should have **_exactly one_** symbol within it … that is to say, no way to mark it as dedicated.  Therefore, if there is ever more than one symbol in such a section, only the first item receives any alignment guarantee beyond it's inherent data type's.

Even a dedicated section is **_fragile_** … it's fine for now, but when maintainers change or teams grow larger, it adds risk because it **_relies on human validation, rather than programmatic validation, of constraints_**.

</details>

<details>
  <summary>Zephyr: An aside...</summary>

<p>

Zephyr may be pertinent to what's common, but is not pertinent as to whether a solution is fragile or not.  Zephyr appears to be fragile for at least the reasons listed above on post-linkage validation.

</details>

In summary, rather than correctness/conciseness/explicit verbosity/forward-compat, this issue is trying to move towards programmatic validation of constraints, to avoid human error validating (or accidental breaking of, such as through ignorance) those constraints.  This appears to be possible today, only via a post-link validation of the binary's sections.

I've decided to close this issue.  Feel free to re-open if you would like to track this for future releases.

closed 2019-07-13 by henrygab

richardtaylorrt, 2019-07-15:
@henrygab - Thank you again for all your thoughtful feedback here.  We'll certainly keep it in mind as this area evolves.