Azure/azure-sphere-samples #59: target API set must be changed to 2+beta 1905 to build

ozhernan, 2019-06-27:
in this readme file https://github.com/Azure/azure-sphere-samples/blob/master/Samples/AzureIoT/README.md

The step below to enable ethernet should include a line to ask the user to change the Target API SET property to 2+beta 1905, otherwise it won't build

4.- In main.c, add the following lines before any other networking calls:

 err = Networking_SetInterfaceState("eth0", true);
 if (err < 0) {
       Log_Debug("Error setting interface state %d\n",errno);
       return -1;
   }

richardtaylorrt, 2019-06-28:
Thanks for raising this.  We'll fix the README as you suggest for this sample, and for other similar samples too.

ozhernan, 2019-06-28:
Thanks, as you point out,  is not consistent between examples, in the HTTPS example, that extra explanation  is not needed since the right API set is configured in the project.

cawhitworth, 2019-07-02:
Fixed in #61

closed 2019-07-02 by cawhitworth