Azure/azure-sphere-samples #60: External MCU update for FW bin about 5MB

BillyHsu, 2019-07-01:
Dear Sir,

We have a customer case is interest AzureSphere OTA function.
But their main chip FW image is about 5MB.
How to make the external FW image update for this kind huge size and go through Azure Sphere Security Service?

Best Regards,
Billy

richardtaylorrt, 2019-07-02:
Hi @BillyHsu - as you say, today, there is not enough room to include such a large image within a customer application on the MT3620.  Please (ask the customer to) raise this feature request [here](https://feedback.azure.com/forums/915433-azure-sphere).

In the meantime, one idea would be for the app to obtain the FW image using a secured connection directly to the customer's cloud.  The Azure Sphere app could obtain the FW image in small pieces, and transfer these incrementally to the other chip.  So the app would never need to contain the entire FW image at the same time.

BillyHsu, 2019-07-09:
Thank Richard,

I will discuss this solution to the customer.

Best Regards,
Billy

closed 2019-07-09 by BillyHsu