Azure/azure-sphere-samples #63: Support ethernet board configuration image for Avnet dev kit

ozhernan, 2019-07-02:
The new Avnet dev kit exposes USI1 (GPIO31-34)and GPIO2 in both mikro bus connectors and there is an ethernet click board that uses the same ENC28J60 part we support
https://www.mikroe.com/eth-click

Are you planning to support a board support image that supports USI1/GPIO2 in additions to the current USI0/GPIO5 we have?

bawilless, 2019-07-03:
I think Oz was referring to ISU1, not USI1 in his request.

We have a couple of PoCs in progress that require the ETH click on an Avnet Starter kit.  A quick turnaround on this request would be appreciated.

Thank you,
Brian

richardtaylorrt, 2019-07-03:
Hi - thanks. I know this is a request that we are already tracking. If you haven't already, please could you also raise this request on our [MSDN forum](https://social.msdn.microsoft.com/Forums/en-US/home?forum=azuresphere) and via your Azure Sphere support representative if you have one. You should expect to get a response via those routes, but please return here if you don't see an update within a week or so.

richardtaylorrt, 2019-08-09:
We're closing this specific issue here, because we understand the overall request is being tracked elsewhere.  (Issues are this repo is for problems with the sample apps that it contains.)

closed 2019-08-09 by richardtaylorrt