Azure/azure-sphere-samples #65: You are not signed in or don't have access to the IoT Central App Capabilities.

EK76, 2019-07-09:
When I run ShowIoTCentralConfig I get this message
You are not signed in or don't have access to the IoT Central App Capabilities.

I have followed the instructions, but I've probably missed something. How do I correct this?

richardtaylorrt, 2019-07-09:
Hi - have you successfully created an IoT Central Application?  When you run the ShowIoTCentralConfig tool does it prompt you to login?  Are you logging in with the same account you use with IoT Central?  Does the login fail with any other messages?

If this doesn't help, it'd be great if you can share screenshots of what you see in IoT Central and what you see when running the ShowIoTCentralConfig tool.  Thanks.

EK76, 2019-07-10:
Hi

Thank you for writing to me.
In my opinion, I have successfully created an IoT Central Application. Here are some screenshots.

[cid:7c5cde70-8c5f-43df-92d3-5a63b5fc51fb]
[cid:8fee6116-8d6e-45ff-bf25-79f5b7c7995a]

Do you wan't more screenshots about my  IoT Central Application?

When I run ShowIoTCentralConfig tool it prompts me to login and I a'm using the same account I use with IoT Central.
The login fail only with the message "You are not signed in or don't have access to the IoT Central App Capablitities"

[cid:d755dd97-edec-4212-892e-021433dc00f5]

Do you need more info?

Best regards,
Ken!

richardtaylorrt, 2019-07-11:
Hi - thanks for the reply.  Unfortunately, I don't see the screenshots.  Please can you try adding them directly to this issue on GitHub, rather than replying via email.  Thanks.

EK76, 2019-07-11:
Hi!

I will do that!

Best regards,
Ken!

EK76, 2019-07-15:
Here is little more info about my issue.
In my opinion I have followed the instruction in the right way. I have created IoT hub and DPS with my live.com account and with the azsphere login commando
I have used the onmicrosoft.com account. Is it done correctly in this way?
[image]

richardtaylorrt, 2019-07-15:
Thanks for the extra information.  I think I can see the problem now.  The "live.com" account is a personal account.  It is not a work/school account.  So you should answer "N" to the first question that the tool asks you.

However, the tool isn't able to work automatically with personal accounts.  So instead, you will see that it helps you gather the information you need manually by copy/pasting it from the IoT Central UI.

(Note that when you answered "Y" to the question, I guess the tool checked if you have any IoT Central accounts associated with your "onmicrosoft.com" account.  It correctly answered that there are not any.)

Please let me know if this helps.

EK76, 2019-07-15:
Now the ShowIoTCentralConfig tool worked for me and I put these values into my app_manifest.json file, 
which can be found in the picture below this text.
[image]
But when I run with Visual Studio it give me the following error.
[image]
I can’t figure it out what causing this problem?

EK76, 2019-07-15:
I have set up a working wifi network connection and Azure Sphere has the latest OS version installed.
[image]
Is there anything else I can check?

richardtaylorrt, 2019-07-15:
Looking at the sample code, one posssibility is that the network isn't "ready" yet.  This only happens when the OS can sucessfully reach an NTP service online to get current time.  So some questions with that in mind:

- Does your network meet [these requirements](https://docs.microsoft.com/en-us/azure-sphere/network/ports-protocols-domains)?
- Does the [HTTPS sample](https://github.com/Azure/azure-sphere-samples/tree/master/Samples/HTTPS/HTTPS_Curl_Easy) work - successfully reading from example.com? 
- Was your device on this network when it updated to 19.06?

EK76, 2019-07-16:
I tried the HTTPS sample but I get this error message.
[image]
[image]
I managed to download the update even if I couldn’t ping to 192.168.0.138, which is Azure Sphere’s 
ip address according to azsphere device show-ota-status
[image]
[image]
I’m using for the moment time.sphere.azure.net. I don’t if it is the correct one?
Perhaps I have something wrong with my NTP service?

richardtaylorrt, 2019-07-16:
Can you try attaching the Azure Sphere device to a different network?  E.g. a mobile hotspot.

Does your network have any restrictions on which devices can connect.  Does it have MAC address filtering, a captive portal, etc?  I.e. a reason why your PC can connect to the internet but your device can't?

It might be an NTP problem in particular, though that's not clear.  I see the requirements also say the network firewall must allow UDP port 123 for NTP.  Does your current network enable this?

Finally, note that the device doesn't respond to pings, so it's expected that these did not work.

EK76, 2019-07-16:
I tried with the mobile hotspot, same error occurred . I allowed UDP port 123 in my firewall and even disabled firewall but it didn't helped. I have not MAC filtering and captive portal enabled or any other restrictions , that I can think of. I can't understand what I missed or done wrong?

richardtaylorrt, 2019-07-16:
Hi - sorry that this is proving so difficult.  Please can you

- run `azsphere get-support-data --output diags.zip` with your device attached, soon after repro'ing the issue
- send the zip file to AZCommunity@microsoft.com with a link to this GitHub issue as well as your subscription ID.

This will allow us to get a support ticket opened for this issue.

floresboy, 2019-07-29:
same issue here
any news on this ?

richardtaylorrt, 2019-07-29:
Hi - @floresboy, this GitHub issue ended up covering multiple different potential problems.  Please could you clarify precisely what you are seeing.
- If it's an issue with running a specific sample app, then please raise a new issue in this repo.
- If it's a general issue in getting started with Azure Sphere, please raise it on our [MSDN forum ](https://social.msdn.microsoft.com/Forums/en-US/home?forum=azuresphere) where our community can help.

Thanks.

richardtaylorrt, 2019-08-09:
We're closing this issue now.  Please re-open here, or on the MSDN forum (as above) if you still need assistance, and we'll be happy to help.  Thanks.

closed 2019-08-09 by richardtaylorrt