Azure/azure-sphere-samples #66: Multiple Ethernet controllers

galvesribeiro, 2019-07-22:
Hello folks!

We're planning on build our device using Azure Sphere and we came to a point where we need to have multiple Ethernet connections on a single Azure Sphere MCU. For this particular case, it will be 9 Ethernet controllers. 

I was walking thru [the docs](https://docs.microsoft.com/en-us/azure-sphere/network/connect-ethernet) and that led me to [this link](https://github.com/Azure/azure-sphere-samples/blob/master/Hardware/mt3620_rdb/EthernetWiring.md) which tells me how to do the wiring for a single ethernet controller. So far so good. Except that we need to have 9 of those.

We know that we can use multiple devices on the SPI but you need to use the CSx (the GPIO) pin to switch to which one we want to read/write data on the SPI. Also, there is the _board configuration image_ that must be built, and I only see docs on how use the `--preset` one, and not how actually _create_ that image.

So, how can I:

1. Attach multiple ethernet connectors to our Azure Sphere-based device
2. Build a custom configuration image ourselves

Thank you! I would really appreciate any help on this.

galvesribeiro, 2019-07-23:
So, on my diggings, I've found this file:

[image]

Which basically is a device tree.

I've reverted it back to a `.dts` and got this:

```dt
/dts-v1/;

/ {

	azconfig {
		version = <0x01>;
		preset = "lan-enc28j60-isu0-int5";
		desc = "Ethernet controller using SPI on ISU0 and interrupt on GPIO5";
	};
};
```

So, I'm assuming the interesting part of this is the `-iso0-int5` to map which SPI (ISU) pin is used and at which GPIO interrupt. 

As [this document](https://github.com/Azure/azure-sphere-samples/blob/master/Hardware/mt3620_rdb/EthernetWiring.md) mentions:

> ENC28J60-H CS: 7 to MT3620 CSA0: Header 2 (lower left) Pin 1

That is _somewhere_ it is hardcoded (or defined by default on the kernel) to expect the GPIO pin 0 as the CS.

Looking at the [linux kernel source (4.9)](https://elixir.bootlin.com/linux/v4.9/source/include/dt-bindings/gpio/gpio.h#L14) it looks like `GPIO_ACTIVE_LOW` would be the correct to select the device...

So, I assume this change _could_ work:

```dt
/dts-v1/;

/ {

	azconfig {
		version = <0x01>;
		preset = "lan-enc28j60-isu0-int5";
		desc = "Ethernet controller using SPI on ISU0 and interrupt on GPIO5";
                cs-gpios = <&gpioa 0 GPIO_ACTIVE_LOW>
	};
};
```

So if I understood correctly, it would be possible to add multiple ethernet controllers as long as we can change the the the `int`, `isu` and also properly set the `cs-gpios` (idk if the value on my example is correct)...

Can someone confirm that? I'm not a linux kernel expert, so I would appreciate if someone could shed some light on it. 

That capability (run multiple ethernet interfaces) is crucial to our project. So it is a go/no-go with Azure Sphere depending on the answer.

Thanks!

richardtaylorrt, 2019-07-23:
Thanks for the question here. To help us investigate further, please could you tell us:
1. What does your device do?  What is the scenario that requires 9 separate Ethernet connectors? Could a switch/hub be used to connect multiple devices to the same network?
2. Are you already working with any Microsoft staff on your project?  (You mention a go/no-go decision.)

Thanks.

galvesribeiro, 2019-07-23:
Hello @richardtaylorrt thanks for the reply.

To answer your questions:

> What does your device do? What is the scenario that requires 9 separate Ethernet connectors? Could a switch/hub be used to connect multiple devices to the same network?

It is a device built by our company (not public yet) with focus on health industry. The reason for the 9 ethernet connectors is because the internal design of the device is basically similar to what you would see on a Dell Blade server chassis. You have blades (in our case cells) on it that are removable and when attached to the backbone of the chassis join the "network". Each of those cells have its individual Azure Sphere MCU and are connected thru USB-C so power and networking will go over this connection. The backbone in other hand, is its own Azure Sphere MCU chip. It will behave as a router for the inter-module communication (there will be a TCP server listening on it, filtering the inter-module messages and routing to the appropriate one without caring about where in the chassis it is connected, after all, it should be an IP on the private network) and to Internet access for the modules thru a LTE controller chip (so they can reach both Azure for OS/App updates and our backend). 

So, problem is that, for this network to work, the backbone MCU must have multiple network controllers (1 per connection/cell). We don't want to have an external network hub but instead to build everything over USB-C as designed.

Also, I've looking closely to the MCU specs, and look that there are only 8 CS pins on it, or at least those are the ones that are mapped on the dev board. If that is the case, we will probably have to reduce the number of modules/cells supported by the product to 8. In the worse case scenario it is acceptable. The major problem now is how to make sure the kernel is switching the CS pins and how we can configure that properly so our backbone can have network for all the modules.

> Are you already working with any Microsoft staff on your project? (You mention a go/no-go decision.)

I'm trying to get in touch with someone. From the chip manufacturer, we've started contact with both Mediatek and AVNET so they can sell us the MCU. In parallel, Jon Morris from IoT team was trying to find someone from MSFT that can help with Azure Sphere last week, but it was Inspire week so they didn't got back yet.

We're all former MSFT FTE, so we know the drill. If you need to contact offline please let me know.

Azure Sphere was actually the reason we decided to build this project the way it is and it is proving to be a fantastic choice. We've just hit that small bump.

galvesribeiro, 2019-07-24:
@richardtaylorrt ?

richardtaylorrt, 2019-07-24:
Hi - this started quite a conversation internally!
- Unfortunately, initial research indicates there's no obvious way you can self-serve to add an additional Ethernet controller with today's platform.  So this would likely be a feature request.  Perhaps you can ask this (copy/link to this trail) on our [User Voice](https://feedback.azure.com/forums/915433-azure-sphere), where we can respond officially.  You could also ask on our [MSDN forum](https://social.msdn.microsoft.com/Forums/en-US/home?forum=azuresphere), where there is further community engagement.  (Issues raised in this GitHub repo are typically for issues in our sample code itself.)
- In theory, you _might_ be able to bring your own Ethernet code directly into your app to run on top of the other SPIs available.  We haven't tested that, but would be very interested if you try it.
- To deal with mutiple CS lines, you might also be interested in [this thread](https://github.com/MicrosoftDocs/azure-sphere-issues/issues/105).

galvesribeiro, 2019-07-24:
@richardtaylorrt thanks for getting back.

> In theory, you might be able to bring your own Ethernet code directly into your app to run on top of the other SPIs available. We haven't tested that, but would be very interested if you try it.

Yeah, that is a possibility. But that also means I have to reinvent the wheel since the OS already provides networking capabilities on it. 

> To deal with mutiple CS lines, you might also be interested in this thread.

That is an alternative but again, we would have to reimplement the ethernet capabilities on top of it.

In our opinion, it should be a transparent setting from the OS. 

That lead us to a path I didn't want to go... To build our own "ethernet switch/hub".

So, assuming that we have the hub built and all the cells are attached to it and we add the backbone Azure Sphere MCU to it as well:

1. Backbone Azure Sphere MCU would setup the DHCP service
2. All the Azure Sphere MCUs on each of the cells would become DHCP clients and receive an IP from the Backbone
3. Backbone IP is also set as the Gateway of the network so the internet traffic from all the cells goes to thru it and are forwarded to the next hop 
4. Local traffic between cells would work, as they are all on the same network
5. Only one ethernet controller exist on each MCU (including the backbone)
6. The LTE controller is attached to Backbone Azure Sphere MCU and it is the internet "router"

With that in mind:

1. Would you think this scenario would work? 
2. Is there any samples on how to configure an Azure Sphere as a Gateway/Router?

Thank you! If that works, at least we have an alternative.

galvesribeiro, 2019-07-26:
@richardtaylorrt can you confirm if that will work?

richardtaylorrt, 2019-07-26:
Hi - overall, we think this would work but it's not something we've tested.
- One challenge is the DHCP service can currently only issue a single IP address to a single client. We'll look to increase this in a future release: please request this via the User Voice link above. In the meantime, you could configure IP statically.
- We don't recommend Azure Sphere acts as a pure gateway/router for downstream unsecured devices, because this essentially gives such devices an unprotected pass-through internet connection.  Accordingly, we normally suggest the Azure Sphere app performs careful business logic to process data sent to/from the cloud. Though in your case, where the downstream device is itself an Azure Sphere device, your security analysis may be different.
- We have a [Private Network Services sample ](https://github.com/Azure/azure-sphere-samples/tree/master/Samples/PrivateNetworkServices) showing how to run an example TCP server on a private local (Ethernet) network.
- We're looking into how someone from the field team can reach out to you to assist further.

galvesribeiro, 2019-07-26:
Hey! Thanks for the reply:

> We have a Private Network Services sample showing how to run an example TCP server on a private local (Ethernet) network.

Yes, that sample is something that the main Azure Sphere MCU would have to provide internal (business) services for the downstream Azure Sphere MCU devices. 

We could also add a DNS service on that main AZ Sphere device so it can provide service discovery as well. 

The only reason we are trying to make the main Az Sphere device a router/gateway, is so the downstream devices can talk to Internet on our backend services and Azure Sphere Cloud. Those devices don't have an LTE chip on it. The only one that does have that is the main Az Sphere one.

> We're looking into how someone from the field team can reach out to you to assist further.

That would be great.

Thank you!

richardtaylorrt, 2019-08-09:
We're closing this issue now.  Please re-open if you still need assistance, and we'll be happy to help.  Thanks.

closed 2019-08-09 by richardtaylorrt

galvesribeiro, 2019-08-09:
@richardtaylorrt Didn't got any return from anyone from MSFT. I've signed the CLA with AVNet earlier this week but nobody got in touch yet. I would appreciate any guidance on that matter. THanks

richardtaylorrt, 2019-08-09:
I followed up.  I heard someone tried to reach out, but had the wrong/missing email address.  Please can you email me directly and I'll connect you: ritaylor at microsoft dot com .  Thanks.

javakys, 2020-07-27:
Hi, 
I'm James from WIZnet in Korea.
We're developing a Dual ethernet Azure Sphere Guardian Device.
One is for the private network and the other for the public network.
We're using W5500 our own hardwired TCP/IP chip for the private network and it works on RT core of MT3620.
You may be able to embed multiple W5500s with sinlge SPI interface and multiple GPIOs as CSs for each W5500.

I don't know you are still concerning this issue because almost a year past.

If you have any question, please contact me. 
My email address is jameskim@wiznet.io

Thank you.