Azure/azure-sphere-samples #67: IoTHubDeviceClient_LL_CreateWithAzureSphereDeviceAuthProvisioning returned 'AZURE_SPHERE_PROV_RESULT_PROV_DEVICE_ERROR'.

gbuckmaster, 2019-07-23:
I've followed the guides verbatim and neither the IOTCentral nor IOTHub guide results in a working scenario.  When using IOTHub, I get the following error.  Having run Wireshark on the network segment in question shows no packet traffic leaving the device at all.  I'm using an MT3620 board.

richardtaylorrt, 2019-07-23:
Hi - this error typically occurs when IoT Central or IoT Hub (DPS) doesn't yet trust the Azure Sphere tenant Certificate Authority.  But it sounds like you have performed the steps to establish this trust?

And this doesn't fit your observation that there is no traffic from the device.  This error should only occur after the device has obtained its certificate, which means that at least some traffic is getting through.

Please can you
- share the full output from the device (Output window in Visual Studio)
- double-check that the device is connected (azsphere device wifi show-status) yet you still see no network traffic at all from the IP address listed there
- try running the [HTTPS sample app ](https://github.com/Azure/azure-sphere-samples/tree/master/Samples/HTTPS/HTTPS_Curl_Easy) and share the output of that too, which will further check for networking issues.

floresboy, 2019-07-30:
will do, thanks!

ramo271285, 2019-08-04:
Hi Richard like floresboy Im using IOTHub and I got the same error 'AZURE_SPHERE_PROV_RESULT_PROV_DEVICE_ERROR'.

I do your steps:
- azsphere device wifi show-status (connected ok) 
- running the HTTPS sample app  (works fine)
- here its my output:

Remote debugging from host 192.168.35.1
IoT Hub/Central Application starting.
Setting Azure Scope ID 0ne0006F727
Opening SAMPLE_BUTTON_1 as input
Opening SAMPLE_BUTTON_2 as input
Opening SAMPLE_LED as output
[Azure IoT] Using HSM cert at /run/daa/dbd3678c-9667-475f-a29d-3ed28bfb2c6d
IoTHubDeviceClient_LL_CreateWithAzureSphereDeviceAuthProvisioning returned 'AZURE_SPHERE_PROV_RESULT_PROV_DEVICE_ERROR'.
ERROR: failure to create IoTHub Handle - will retry in 60 seconds.
IoTHubDeviceClient_LL_CreateWithAzureSphereDeviceAuthProvisioning returned 'AZURE_SPHERE_PROV_RESULT_PROV_DEVICE_ERROR'.
ERROR: failure to create IoTHub Handle - will retry in 120 seconds.

Please can you help me here

Regards

Gaaaabor, 2019-08-09:
Hi ramo271285,

I had the same issue, make sure that in the "app_manifest.json" for the "AllowedConnections" you add your IoT Hub's link but **do not delete the existing item**.

First time I replaced the global link with mine, it was not working, then I added the again global one (which was already there in the example) and it worked.

ramo271285, 2019-08-10:
Hi Gaaabor,

It was that, now it works!

Thank you very much
Best regards

closed 2019-10-03 by richardtaylorrt