Azure/azure-sphere-samples #68: Xplat Azure Sphere CLI/Tools

galvesribeiro, 2019-07-24:
Hello folks!

I've looking on how the tooling for Azure tooling works and it comes to:

1. The `net461` `azsphere` CLI
2. Azure Sphere Device Comms Service: `C:\Program Files (x86)\Azure Sphere Device Communication Service`
3. Azure Sphere SDK (toolchain, sysroot): `C:\Program Files (x86)\Microsoft Azure Sphere SDK`
4. Some VS Extensions

The CLI has no reason it could be turned it `netcoreapp2.x` as it doesn't have native dependencies on the OS that I've found out.

The SDK is basically an `arm-poky-linux-musleabi` toolchain with its sysroot, so it can be easily built for other OSes like Linux/OSX while the toolchain is the same.

 The Comms service is an asp.net core app bridging the CLI commands to interact with the device. This one have a few PInvokes on the Win32 APIs that could be easily wrapped in a simple abstraction and implement the other OS calls easily.

The VS extensions could be easily made VSCode extensions an much of what is happening on it can basically use the existent GDB debugger from VSCode to allow debugging.

So, can someone tell me if there is a plan to shortly make the SDK cross-platform? The majority of IoT companies don't use Windows for development and given that VSCode and specially .Net Core are presented so mature, people could really use the support.

If the SDK were OSS'd (everything but the VS extensions like in other projects), I would contribute the port myself. Otherwise, soon the community will come with unofficial ports that would work, but is not something companies would use as it is not stamped by MSFT.

Thanks! Keep the great work!

richardtaylorrt, 2019-07-31:
Hi - sorry for slow response.  We've certainly heard this feedback!  See links below, which you might want to upvote.  I hope we'll have an update to share on UserVoice sooon.
https://feedback.azure.com/forums/915433-azure-sphere/suggestions/35567668-add-mac-vs-code-support
https://feedback.azure.com/forums/915433-azure-sphere/suggestions/35968183-support-linux-distros

richardtaylorrt, 2019-08-09:
We're closing this issue now.  Please re-open if you still need assistance, and we'll be happy to help.  Thanks.

closed 2019-08-09 by richardtaylorrt