Azure/azure-sphere-samples #69: List of compatible peripheral devices

TomaszWaszczyk, 2019-07-25:
There is any list of peripherals like oled display which is compatible with Azure Sphere?

richardtaylorrt, 2019-07-31:
Hi - sorry for slow response.  We don't have an official list right now.  But in general, you should be able to drive any display or peripheral over I2C, SPI or UART if it doesn't have demanding size or peformance requirements.

Do you have a particular display in mind?  We'd be very interested in your experiences in porting its driver code to Azure Sphere.

Over time, we hope to help support the building out of an ecosystem so the community can find code for running all sorts of different peripherals on Azure Sphere.

richardtaylorrt, 2019-08-09:
We're closing this issue now.  Please re-open if you still need assistance, or ask a question in our [MSDN forum](https://social.msdn.microsoft.com/Forums/en-US/home?forum=azuresphere), and we'll be happy to help.  Thanks.

closed 2019-08-09 by richardtaylorrt