Azure/azure-sphere-samples #70: Unable to receive data from RTApp to High Level Application (Intercomms sample)

KomalBagai, 2019-07-30:
I am unable to receive data from my RT Application to High Level application. The High level application keeps on sending hello world until I receive this error in RT App

///
openocd: Error: libusb_handle_events() failed with LIBUSB_ERROR_IO
openocd: Error: unable to purge ftdi rx buffers: LIBUSB_ERROR_IO
openocd: Error: MPSSE failed
///

how can I resolve this?

richardtaylorrt, 2019-07-30:
Hi - are you able to run and debug any RT apps at all?  Have you been able to follow the [QuickStart ](https://docs.microsoft.com/en-us/azure-sphere/quickstarts/qs-real-time-application)?  Specifically:

- Have you successfully run `azsphere device prep-debug --enablertcoredebugging`?
- Are you using Visual Studio to deploy and debug?
- Can run and set a breakpoint in the [Hello World RT app](https://github.com/Azure/azure-sphere-samples/tree/master/Samples/HelloWorld/HelloWorld_RTApp_MT3620_BareMetal)?
- Are you able to see any of its [UART log output](https://docs.microsoft.com/en-us/azure-sphere/quickstarts/qs-real-time-application#set-up-hardware-to-display-output)?

If this is all working, then you can go back to the 'inter-core' RT app and use the debugging and logging to determine if it is receiving messages from the high-level app.

KomalBagai, 2019-07-30:
> Hi - are you able to run and debug any RT apps at all? Have you been able to follow the [QuickStart ](https://docs.microsoft.com/en-us/azure-sphere/quickstarts/qs-real-time-application)? [...]

Yes, it is all working, I am using visual studio itself and the intercore sample is also working and is able to receive messages. But when i try to interface it with the azure iot services, I am unable to receive messages from the RTapp. The error stated above was eventually resolved by updating the os. But I am still uanble to receive the meassages from the RTapp to Azure iot interfaced high level app. Is there any visual studio configuration that I have to do before communicating to an another?

richardtaylorrt, 2019-07-30:
So the problem is that the high-level app doesn't receive anything from the RT app? 

It sounds like you may have changed the high-level app (to add Azure IoT connection).  This may have ressulted in a new 'component ID' in its app manifest.  This new component ID for the high-level needs to be placed in the "AllowedApplicationConnections" of the RT app's manifest (see [here](https://docs.microsoft.com/en-us/azure-sphere/app-development/inter-app-communication#application-manifest-settings)) and in the "partnerComponents" of the RT app's launch.vs.json (see [here](https://docs.microsoft.com/en-us/azure-sphere/app-development/sideload-app#sideload-more-than-one-application)).

If this doesn't work, please share the log debug output from both apps (Device Output window in Visual Studio for high-level app, and serial connection terminal output for RT app).  Thanks.

KomalBagai, 2019-07-31:
I would rather reframe the problem that AzureIoT High Level Application is not able to communicate with the RT app, since the RT app is neither able to send nor receive the "helloworld" as in the intercore sample.
Component ID for my RT app: "005180bc-402f-4cb3-a662-72937dbcde47"
Component ID for my High Level app: "819255ff-8640-41fd-aea7-f85d34c491d5"

As in the serial connection terminal output from RT app below you can see that its trying to receive data from the RT app but isnt getting anything hence printing nothing

```
 Received message of 35bytes:
  Component Id (16 bytes): 819255ff-8640-41fd-aea7-f85d34c491d5
  Reserved (4 bytes): 00280003
  Payload (15 bytes as hex): 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
  Payload (15 bytes as text): 
Received message of 35bytes:
  Component Id (16 bytes): 819255ff-8640-41fd-aea7-f85d34c491d5
  Reserved (4 bytes): 00280003
  Payload (15 bytes as hex): 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
  Payload (15 bytes as text): 
Received message of 35bytes:
  Component Id (16 bytes): 819255ff-8640-41fd-aea7-f85d34c491d5
  Reserved (4 bytes): 00280003
  Payload (15 bytes as hex): 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
  Payload (15 bytes as text): 
Received message of 35bytes:
  Component Id (16 bytes): 819255ff-8640-41fd-aea7-f85d34c491d5
  Reserved (4 bytes): 00280003
  Payload (15 bytes as hex): 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
  Payload (15 bytes as text): 

```

On the other hand my Azure IoT High Level application keeps on sending "hello world" to the RT app and telemetry to the cloud (as in the sample)

```
Remote debugging from host 192.168.35.1
IoT Hub/Central Application starting.
Setting Azure Scope ID 0ne00062975
Opening SAMPLE_BUTTON_1 as input
Opening SAMPLE_BUTTON_2 as input
Opening SAMPLE_LED as output
sucesssSending: Hello-World-0
Sending: Hello-World-1
Sending: Hello-World-2
Sending: Hello-World-3
Sending: Hello-World-4
Sending: Hello-World-5
[Azure IoT] Using HSM cert at /run/daa/41669393-a4bb-4f47-92c9-b8842d46c9af
IoTHubDeviceClient_LL_CreateWithAzureSphereDeviceAuthProvisioning returned 'AZURE_SPHERE_PROV_RESULT_OK'.
Sending IoT Hub Message: { "Temperature": "30.90" }
INFO: IoTHubClient accepted the message for delivery
Sending: Hello-World-6
Sending: Hello-World-7
Sending: Hello-World-8
Sending: Hello-World-9
Sending: Hello-World-10
Sending: Hello-World-11
Sending IoT Hub Message: { "Temperature": "31.50" }
INFO: IoTHubClient accepted the message for delivery
IoT Hub Authenticated: IOTHUB_CLIENT_CONNECTION_OK
Sending: Hello-World-12
Sending: Hello-World-13
Sending: Hello-World-14
Sending: Hello-World-15
Sending: Hello-World-16
Sending IoT Hub Message: { "Temperature": "31.60" }
INFO: IoTHubClient accepted the message for delivery
Sending: Hello-World-17
```
I guess that there is some property setting problem in my high level application. I have already done the following changes:
1. Put the RTapp component ID in the code (on which a communication socket is created)
2. Entered the RTapp component ID in app-manifest.json file under "AllowedApplicationConnections"
3. Entered the RTapp component ID in the project properties>debugging>partner components.

In the RT app, I have done the following changes: 
1. Put the High level app component ID in app_manifest.json file under "AllowedApplicationConnections"
2. Entered the High level app component ID in launch.vs.json file under "partner components"

What more changes are to be done in the applications?
Or is there any way of connecting my RTapp to the azure IoT services (hub in my case)?

closed 2019-08-05 by KomalBagai

richardtaylorrt, 2019-08-05:
[Resolved via private messaging]