Azure/azure-sphere-samples #71: Factory RESET

ddizzypgh, 2019-07-31:
what is the key sequence to factory reset the device?

Don't send me a link for the azsphere client.

If the device is in debug mode.... you can't install it on another device,

The software fails with: Product or related product not found: {D0C37DF0-56F6-4D80-BDFC-74EEB20F578D}

Please, do not send me to some dumb link.

My question is clear: how do you factory reset the device without software>

as in, "hold user button1 and reset", boot the device, wait ten seconds and the led will blink 4 times.

seriously, My question is clear

richardtaylorrt, 2019-07-31:
There is no phyiscal sequence to factory reset the MT3620 dev kits - this is an aspect of the security story.  To get back to initial state you can 'recover' (erase and re-image) the device: `azsphere device recover`.  Then you can `azsphere device prep-debug` and re-deploy the 'blink' app from Visual Studio.  I know you don't want links(!) but essentially, this means these steps: [Update the OS](https://docs.microsoft.com/en-us/azure-sphere/install/install-os) and [Quick Start](https://docs.microsoft.com/en-us/azure-sphere/quickstarts/qs-blink-application).

If this doesn't helpo, please could you let us know:

- Which MT3620 dev kit are you using?  Is it from Seeed or Avnet?
- What you mean by "you can't install it on another device"?
- Which command or process fails with the "Product or related product not found" message?

Thanks.

ddizzypgh, 2019-08-04:
this is the most

ddizzypgh, 2019-08-04:
So, I can't put a reset pin/button/Capacitive sensing zone into the design
to reset the firmware??????

Again, I cannot get the Azure SPHERE SDK to install on my Microsoft
Certified Windows 10 Enterprise (using a DC in Azure no less), with
certified Visual Studio 2017 Professional and certified Visual Studio 2019
Professional installed.

So, this device is bricked until you guys fix the installer? Your people
keep closing this thread.... however, it is a real thing and has NOT been
resolved.

Before I get started: you said, "this is an aspect of the security story.
To get back to initial state you can 'recover' (erase and re-image) the
device: azsphere device recover"

So, you are saying any person that can download the azure sphere command
line utility can "recover" the device... and re-image..... and I am
laughing, because you said this makes the device secure?

I guess you are right if you are me.....today

This is the most ridiculous reply, in the history of my career....... and
of any reply, about any devices..... except for the zune....however.... it
is the same reply and the same mistake....

So, you are legit telling me.... *cough*.....

I build 1,000,000 devices with firmware I compiled.

Seal the azure sphere in a container or enclosure of some sorts, ship them
to customers...... that use them, in a production environment...... all
over the world

and if something goes wrong...... I can't put a reset pin/button/Capacitive
sensing zone into the design to reset the firmware??????

I have to tell my customers to ship the 1,000,000 devices back,
re-provision them or they have to download the azure sphere command line
utility???

No thanks, I would rather use an ESP8266-01, which btw costs about 32
cents, and the Azure Iot c++ Libraries fit on it if you are crafty......and
can reset the firmware, with no issue by adding any kind of user
buttons/capactitive touch etc.

what if a mistake was made with the networking piece and can't get online?
What if my design Doesn't EVEN IMPLEMENT WIFI and I utilize the BLE port
that is built in?

how can apps update the firmware if it can't be reset/written to?

Man,,,, I am always the dumbest sucker for the crap you guys push.....

I have windows 10 enterprise, visual studio 2017 and 2019 with the latest
updates.

I urge you to install a VM and install windows 10 enterprise, visual studio
2017 (latest updates) and visual studio 2019 (latest updates), then install
the latest Azure Sphere SDK

watch when you get the error "0x80131500 unspecified" error, check the log
file and it says: Product or related product not found:
{D0C37DF0-56F6-4D80-BDFC-74EEB20F578D}

watch when it fails

here is the google search:
https://www.google.com/search?q=D0C37DF0-56F6-4D80-BDFC-74EEB20F578D&rlz=1C1GCEJ_enUS859US859&oq=D0C37DF0-56F6-4D80-BDFC-74EEB20F578D&aqs=chrome..69i57.399j0j4&sourceid=chrome&ie=UTF-8

I attached a screen shot of it failing in FRONT of this email I just typed
just to show how ridiculous this is.

If AN AZURE SPHERE DEVICE CANNOT BE RESET TO FACTORY DEFAULTED FIRMWARE BY
USER INTERVENTION THERE IS NO CHANCE ANYONE WILL USE THIS PLATFORM

PERIOD.

This has been an ancient practice that I cannot believe was overlooked.
yikes.... who is the lead engineer on this one?

The MCU chip from the BOM docs, used on both versions of the azure sphere
boards, support this feature

Simple as this: UPDATE THE FIRMWARE, button1 and button2 pressed together
on power up for X seconds, take the rom from sector A and move it to sector
C.

I can tell you now, the fact that there are 2 user input buttons.... on the
dev kit..... and when everyone finds out....they are just to "press" and
you can't physically reset the firmware...... this is going to be the funny
story that engineers, like myself, will sit around and joke about a year
from now. We are already laughing about the term "iot"

It's also ridiculous that the documentation says the the azure sphere
command client will only be supported until the EOL determined by Microsoft.

So... cool... my new fancy device you installed at your house. Microsoft
comes out with Winblows 13, which no longer supports the azure sphere
command line utility

Don't get mad when you see me or other people posting about this nonsense.

This device is a doorstop within 4 months and the "azure sphere device" is
a waste of time.

Attached is my screenshot.

=D=

MohitGargMSFT, 2019-08-05:
@ddizzypgh Apologies for the inconvenience you are currently facing with the service. It seems that it would be best to work with dedicated engineer to address a few issues been reported above. Please open an [Azure Support Ticket](htps://aka.ms/azsupt) so that a dedicated Azure Support Engineer could further check on your scenario. 

In case you have limitation in your support plan to open support ticket let us know and we will work to enable one time free support on your subscription to get you unblocked. Thanks for using our service and we look forward to provide all the possible support.

richardtaylorrt, 2019-08-09:
We're closing this issue now.  Please raise a support ticket, as suggested above, or let us know if you are unable to do so.

closed 2019-08-09 by richardtaylorrt