Azure/azure-sphere-samples #75: C++ Support

saltyJeff, 2019-08-05:
I have created a version of the blink demo with C++ support if anyone wants to add it.

[https://github.com/saltyJeff/AzureSphereCpp/blob/master/AzureSphereBlink1/main.cpp](https://github.com/saltyJeff/AzureSphereCpp/blob/master/AzureSphereBlink1/main.cpp)

It does involve ~~statically linking~~ copy pasting files against a C++ stdlib implementation, but that also gives you the flexibility to cut out the STL containers you don't want.

The modified stdlib implementation I used is here:
[https://github.com/saltyJeff/uClibcpp](https://github.com/saltyJeff/uClibcpp)

richardtaylorrt, 2019-08-09:
Thanks for sharing this with us, very interesting.  We're exploring how we can more formally support aspects of C++ with Azure Sphere.  If you haven't already, please feel free to share this on the [MSDN forum](https://social.msdn.microsoft.com/Forums/en-US/home?forum=azuresphere) too.

closed 2019-10-03 by richardtaylorrt

maskimthedog, 2020-01-24:
We just had Azure Sphere training earlier this week and my coworker found Jeff's uClibcpp since we desire to utilize C++ on the Sphere.

mehrdadhe, 2020-05-25:
@richardtaylorrt is there any plan to support CPP in future?
Thanks