Azure/azure-sphere-samples #76: System Time and NTP

jclasquin, 2019-08-06:
Hello,

I'm looking at the SystemTime example and I see the note saying the system time may be overwritten by NTP if the device is connected to the Internet. 

I actually WANT the system time to be overwritten by the network time, but that's not happening.  I am connected to the Internet but my time always shows May 1935. 

Is there something I need to do to make NTP work?

Thanks,
Joel

richardtaylorrt, 2019-08-09:
NTP should work if the device is connected to a network that meets [these requirements](https://docs.microsoft.com/en-us/azure-sphere/network/ports-protocols-domains).  Please let us know if this isn't what you're seeing.

In particular, are you able to run the [HTTPS sample](https://github.com/Azure/azure-sphere-samples/tree/master/Samples/HTTPS/HTTPS_Curl_Easy)?  If not, please share the output from both these samples.  Thanks.

richardtaylorrt, 2019-10-03:
Hi - we're closing this issue now.  Please let us know if we can help further.

closed 2019-10-03 by richardtaylorrt