Azure/azure-sphere-samples #77: Azure - PlatformIO

Wiz-IO, 2019-08-07:
https://github.com/Wiz-IO/platform-azure
if you are interested... need litle help

Wiz-IO, 2019-08-08:
Question:
How to **manually** config **TAP-Windows Adapter V9** to work with Asure Sphere board

all other work ( from upper link ) compilng, image maping... need only deploy
please help

richardtaylorrt, 2019-08-09:
Hi - does this [troubleshooting advice for the TAP adapter](https://docs.microsoft.com/en-us/azure-sphere/install/troubleshoot-installation#check-the-tap-windows-adapter-configuration) help?  If not, please could you explain in a little more detail what problem you are seeing, and what you expect to happen instead?  Thanks.

Wiz-IO, 2019-08-09:
The "problem" is:
**New PC, no VS, no Sphere SDK** ... 
VS Code + PlatformIO + my port for Sphere
port have: compiler from sdk (arm-poky-linux-musleabi-hf)
tools as azsphere.exe and uuidgen-64.exe
ALL this work ... HighLevelApp Linux compilations and now Arduino :)

but I cannot "install/enable" Azure Shpere Device Communication Service WITHOUT original sphere SDK
original SDK is 3Gb VS+SDK
my port is only 150Mb

Wiz-IO, 2019-08-09:
this is how it works on a computer with installed original SDK
... Arduino ...
https://www.youtube.com/watch?v=bPYGXtNt8fg

richardtaylorrt, 2019-08-09:
Hi - please can you ask this question on the [MSDN forum](https://social.msdn.microsoft.com/Forums/en-US/home?forum=azuresphere).  Issues raised here are really only for problems with our sample apps.

It's great to see what you're trying to build.  This may not be something we can officially support, but members of the community on MSDN may be able to help you.  It will help if you can specify precisely what information you need to know from them.  (One piece of information: we configure the TAP driver with a static IP config of 192.168.35.1/255.255.255.0 .)

Wiz-IO, 2019-08-09:
size correction - zip file is only 56M :)

I know... I ping 192.168.35.1 without problems

Pinging **192.168.35.1** with 32 bytes of data:
Reply from 192.168.35.1: bytes=32 time<1ms TTL=128
Reply from 192.168.35.1: bytes=32 time<1ms TTL=128

Pinging **192.168.35.2** with 32 bytes of data:
Reply from 192.168.35.1: Destination host unreachable.
Request timed out.
Request timed out.

maybe **Azure Shpere Device Communication Service** not work with UART

Wiz-IO, 2019-08-09:
any solutions how to install/set ONLY Azure Shpere Device Communication Service
[image]

richardtaylorrt, 2019-08-09:
Hi - unfortunately, as I say, I don't think this isn't something we will officially support.  Though you are welcome to ask this question on the [MSDN forum](https://social.msdn.microsoft.com/Forums/en-US/home?forum=azuresphere) to see if the community can help. Issues raised against this repo are really only for problems with the sample apps that it contains.

Wiz-IO, 2019-08-09:
ok ... close

closed 2019-08-09 by richardtaylorrt