Azure/azure-sphere-samples #79: RTApp: Example of GPT3 use in interrupt mode

jgroman, 2019-08-11:
Azure Sphere documentation [mentions](https://docs.microsoft.com/cs-cz/azure-sphere/app-development/use-peripherals-rt) that GPT3 timer is interrupt-based and offers the highest available resolution and recommends its use.
Would it be possible to extend [GPIO_RTApp_MT3620_BareMetal sample](https://github.com/Azure/azure-sphere-samples/tree/master/Samples/GPIO) to demonstrate GPT3 use in interrupt mode as well? 
The only sample using GPT3 [ADC_RTApp_MT3620_BareMetal sample](https://github.com/Azure/azure-sphere-samples/tree/master/Samples/ADC) implements simple blocking wait loop only on this timer.

richardtaylorrt, 2019-10-03:
Thanks for the suggestion.  We've added it to our backlog for consideration.

richardtaylorrt, 2019-11-26:
Hi @jgroman  - Mediatek have recently released [additional drivers and samples ](https://github.com/MediaTek-Labs/mt3620_m4_software) for the MT3620 RT cores.  In particular, I see [this ](https://github.com/MediaTek-Labs/mt3620_m4_software/blob/master/MT3620_M4_Sample_Code/BareMetal/OS_HAL/inc/os_hal_gpt.h) about the GPTs.  Hope this helps.  Please do raise further questions and suggestions on that repo.  Thanks.

closed 2019-11-27 by jgroman