Azure/azure-sphere-samples #81: Nrf52 Sample help

jflynn129, 2019-08-21:
When you compile this sample and the WindowsApp and try to get it to work, it is a hit or miss proposition.  It appears that the problem is that Windows10 automatically pairs with the Nrf52 board so when you run the WinodwsApp in the Sample, it fails to connect.  Sometimes, you can go into the windows bluetooth and manually disconnect it so it isn't paired and it will work (but this appears to be a hit or miss and thus there is no consistency in operation).  When the WindowsApp from the Sample is able to capture the device, it works as documented, but the problem is how to get it to consistently work when using Windows 10.

Can you provide a sequence of steps to get the WindowsApp to consistently connect with the Nrf52 board such that the Windows10 Bluetooth application doesn't interfere?

jflynn129, 2019-08-21:
BTW, a suggestion is to Allow Windows10 Bluetooth application to pair with the Nrf52, enter the passkey, etc.  After Windows has paired with the Nrf52 board, start the WindowsApp in the Sample. Rather than have it perform the connection/pairing, have it simply ensure it is connected, if it is, then perform all the other standard activities.  The problem seems to be a race condition between the sample WindowsApp that is trying to pair with the Nrf52 and Windows10 which is automatically pairing with the Nrf52.  Once Windows10 is paired, the sample WindowsApp will never work.

jflynn129, 2019-08-21:
I would modify the sample WindowsApp myself, but I've never done anything in Windows C#

richardtaylorrt, 2019-08-22:
@jflynn129 - thanks for the feedback, we're looking into this.

dafang982, 2019-08-22:
Hi @jflynn129, have you tried to delete the pairing for the nRF52 in your Windows Bluetooth settings so that you can create a new bond? (As per documented at [here](https://github.com/Azure/azure-sphere-samples/tree/master/Samples/WifiSetupAndDeviceControlViaBle#manage-known-companion-devices))

If not, please do that as shown below:
[image]

The reason you have to do this manually is that once the pairing is removed from the nRF52 (by pressing and holding the Button A on Azure Sphere device), the Windows app can no longer establish connection with the nRF52 device using the previous BLE paring key.

Please let me know whether this solves your problem. If not, could you provide a step by step guide on how to reproduce the issue from a brand new setup?

jflynn129, 2019-08-22:
yes, I did that to get it working. the problem is when you start the application.  To get any consistency, each time, you must delete all bonds in the Sphere device (first), and remove the pairing in Windows 'Bluetooth and other devies' (second), then use the  'Add Bluetooth or other device' function in 'Bluetooth & other devices' to add a new pairing.  If you try and use the WindowsApp that comes in the Sample repository, it doesn't work because Windows pairs with it before the WindowsApp can and it never works.   So I went in and modified the app as I previously described to get it to work.  After I modified it, I have a repeatable scenario I can use.

I worked through the problem you describe (where the sphere has a established pairing), but the WinodowsApp always wants to try and reconnect when you hit "connect" even though it has already been connected via Windows.  Because of this, the WindowsApp is useless without any modifications--At least that is my experience.

dafang982, 2019-08-27:
Once the two BLE devices are paired, the next time you start the Windows app it can connect to nRF52 without re-initializing the pair process again, unless you have deleted the bond from nRF52 device by pressing and holding the Azure Sphere button.

I'm not sure I understand how the device is already connected via Windows before you hit the "Connect" button in the Windows application. Once the Windows application exits, the nRF52 device will change from "Connected" state to "Paired" state after a few seconds (as shown in the screenshot from my previous post). As nRF52 won't trigger the connection itself, it should remain in that state until you start the Windows application again and click on the "Connect" button.

dafang982, 2019-08-28:
Hi @jflynn129, thank you for the extra information you provided. This has allowed us to reproduce two similar scenarios which we believe may also address your issue:

1. After pairing, if the BLE bond is removed from Windows, the bond still exists on the nRF52 device.
When the Windows application tries to re-connect to the nRF52 side to start a new pairing process, the nRF52 will reject it as it believes it has already paired with this Windows BLE device. Unfortunately, this is a BLE limitation. To mitigate this you can press and hold button A on the Azure Sphere device for 3 seconds (when it has the sample application installed and running), which will result in removing all existing bonds on the nRF52 device. The nRF52 device will then accept any subsequent pairing request coming from the Windows application, resolving this issue.

2. The Windows application can fail to connect to nRF52 device if the application on the Azure Sphere device is restarted while the Windows application is running and has an ongoing BLE connection with nRF52 device. 
This is a limitation in the caching implementation of the Windows BLE stack, as the cached BLE connection isn’t refreshed in this scenario. You can mitigate this issue by restarting the Windows sample application, finding the nRF52 BLE device, and connecting to it. 
Like you, we value the importance of clear high-quality documentation, and thank you for bringing your issues to our attention regarding this. We will address the documentation of the above scenarios in a future update.

richardtaylorrt, 2019-10-03:
Hi - we're closing this issue now.  Please let us know if we can help further.

closed 2019-10-03 by richardtaylorrt