Azure/azure-sphere-samples #84: Network is not up using wired-ethernet

matsujirushi, 2019-09-03:
modified main.c and app_manifest.json of HTTPS_Curl_Easy, but it got error message.
"Not doing download because network is not up."
Could you tell me how to fix?

**Setup:**  
1. azsphere recover
2. azsphere image package-board-config --preset lan-enc28j60-isu0-int5 --output enc28j60-isu0-int5.imagepackage
3. azsphere device prep-debug
4. azsphere device sideload deploy --imagepackage enc28j60-isu0-int5.imagepackage

**Application:**  
1. clone https://github.com/Azure/azure-sphere-samples/tree/master/Samples/HTTPS/HTTPS_Curl_Easy
2. Modify main.c and app_manifest.json
[image]
[image]

**Output:**
```
Remote debugging from host 192.168.35.1
cURL easy interface based application starting.
This sample periodically attempts to download a webpage, using curl's 'easy' API.
Not doing download because network is not up.

Not doing download because network is not up.

Not doing download because network is not up.

Not doing download because network is not up.
```

FYI
https://social.msdn.microsoft.com/Forums/en-US/8b07758e-08c3-4e04-ac70-e3afcd6d42a7/network-is-not-up-using-wiredethernet?forum=azuresphere

richardtaylorrt, 2019-09-04:
Hi - please could you tell us more about the Ethernet network the device is connected to? Is it assigning the device an IP address? Does it have a DNS server? Does it meet the network requirements explained [here](https://docs.microsoft.com/en-us/azure-sphere/network/ports-protocols-domains)?

It may help to call Networking_GetInterfaceConnectionStatus() to get more information about the network status.

Alberto-Vega-MSFT, 2019-09-06:
Hi @matsujirushi would you please provide the info Richard is requesting?

matsujirushi, 2019-09-10:
Sorry for replying late.

My network changed, Wired-Ethernet is working.

case 1:
Internet <-(PPPoE)-> Wi-Fi Router A(w DHCP/DNS) <-(Wi-Fi)-> Azure Sphere
case 2:
Internet <-(PPPoE)-> Wi-Fi Router A(w DHCP/DNS) <-(Wired)-> Azure Sphere
case 3:
Internet <-(PPPoE)-> Wi-Fi Router A(w DHCP/DNS) <-(Wired)-> Router B(w DHCP/DNS) <-(Wired)-> Azure Sphere

case 1 and 2 is NG.
case 3 is fine. (Add subnet)

I will close this issue.
(And I plan to investigate the cause of failure in case 1 and 2.)

Thanks.

matsujirushi, 2019-09-12:
I got network capture.
AzureSphere don't sync time server of case 1 and 2.

[image]

Azure Sphere ask time to name server (!)

[image]

And the name server not respond time.

```
matsujirushi@wioserver:~$ sudo ntpdate -q 95.100.173.129
server 95.100.173.129, stratum 0, offset 0.000000, delay 0.00000
12 Sep 05:35:17 ntpdate[1006]: no server suitable for synchronization found
```

richardtaylorrt, 2019-09-17:
Thanks for raising this - it was very useful.  Our team has investigated and have identified the problem.  We will fix in an upcoming OS release.

richardtaylorrt, 2019-10-02:
We expect this problem is fixed in our 19.09 release.  This release is [currently available in our Release Evaluation feed](https://azure.microsoft.com/en-gb/updates/update-19-09-for-azure-sphere-public-preview-now-available-for-evaluation/).  If you are able, please set up a device to access this feed and let us know if it fixes the problem. Otherwise, this release should hit the main Retail feed soon.

matsujirushi, 2019-10-03:
Thanks for your support.
I tested.

```
C:\>azsphere device show-ota-status
Your device is running Azure Sphere OS version 19.09.
The Azure Sphere Security Service is targeting this device with Azure Sphere OS version 19.09.
Your device has the expected version of the Azure Sphere OS: 19.09.
Command completed successfully in 00:00:06.7638308.

C:\>
```

[image]

closed 2019-10-03 by matsujirushi