Azure/azure-sphere-samples #85: Sending data from a RT application to a High-Level one

persello, 2019-09-04:
Hello,
I built a simple M4 application which acts as a software serial receiver. The receiver logic works fine, in fact I'm able to send the received characters over the debug UART.
The data received from this application should be sent to a High-Level application, and this should happen as soon as new data is available.
In the IntercoreComms sample, I can see that the bare metal app sends back the message it received from the high-level one after editing it.
I've seen that before the actual message there are 20 bytes, 16 for the component ID and 4 that are reserved. This part before the payload is left as it is by the M4 application. My question is, is it possible to have a bare metal app send a message to a high-level one without needing the last one to send continuously "empty" messages?
Are the reserved bytes constant or calculable in some way?
I've tried to send a character from the real time app by using a reserved byte sequence I've seen in another issue.

This is the non-working part of the M4 app, the `EnqueueData` and the UART-related functions are taken from the mt3620-intercore.c/h and mt3620-uart-poll.c/h files used in the IntercoreComms sample.
``` c
// 16: Component ID, 4: reserved, 1: character.
char data[16 + 4 + 1] = { /* Component ID of High-Level appn: */ 0x4e, 0xf2, 0xce, 0xfc, 0x2c, 0x95, 0x41, 0x8a, 0x98, 0x7e, 0x3f, 0xda, 0x13, 0x3f, 0xd6, 0x63, /* Reserved: */ 0x00, 0x28, 0x00, 0x03, /* Data: */ 0x00 };

// This is the byte I want to send
data[20] = serial.lastCharRead;
EnqueueData(inbound, outbound, sharedBufSize, &data, 21);

// I can correctly read the byte on the TX line
Uart_WriteStringPoll(&serial.lastCharRead);
```

This is the High-Level app implementation:

```c
const char* lowLevelAppId = "005180bc-402f-4cb3-a662-72937dbcde47";
int sockfd = 0;

int initSSSocket() {
	sockfd = Application_Socket(lowLevelAppId);

	if (sockfd == -1) {
		Log_Debug("INTERCORECOMM: Error while opening socket to M4 RT application. Error: \"%s\" (%d).\n", strerror(errno), errno);
		return -1;
	}

	return 0;
}

int readStringSS(char** data) {

	// Stop when empty or on \r\n
	char c;
	int result;
	do {
		result = recv(sockfd, &c, 1, MSG_DONTWAIT) > 0;

		// We received a character
		if (result)
			strncat(*data, &c, 1);

		// Break on \r\n
		char* ending = strrchr(*data, '\r');
		if (ending && !strcmp(*data, "\r\n")) break;
	} while (result > 0);

	if (strlen(*data)) {
		Log_Debug("INTERCORECOMM: New data (%s).\n", *data);
		return 0;
	}
	else {
		return -1;
	}
}
```

In the High-Level app the initialization succeeds and the `initSSSocket()` function returns 0.
Did I do something wrong? Are there any other resources on the inter-core communication process? On the Azure Sphere documentation there is not much information on the message structure other than
> The message format is similar to a datagram.

Do the reserved bytes contain length or other variable information?

Thanks,
Riccardo

persello, 2019-09-04:
Just realized that the first three Component ID's parts are little endian. I will retry now by applying this change.

persello, 2019-09-04:
Sorry for opening, I'll leave this for reference. Reversing the byte order of the first three blocks (4 bytes, 2 bytes and 2 bytes) of the component ID actually works. As reserved bytes I used `0x00, 0x28, 0x00, 0x03`.

nishithvpoojary, 2021-03-12:
Hi Persello.
I was also trying to send the data from RTapp to HLApp without getting any message from the HLApp. I removed the sending functions in the HLApp. But now the HLApp doesn't show any output. Please help me with this. I didn't understand what you meant by reversing the bytes? Would you please elaborate? Thank You. Anyone else who knows about this, please help me.

persello, 2021-03-12:
If I remember correctly, I wrote the UUID (as bytes) in the same order as you read it. In reality some implementations of UUID encodings use a mixed-median format, especially in Microsoft software. I don't know if things changed with an update, but you should try to encode the Component ID this way (from Wikipedia):

> For example, `00112233-4455-6677-8899-aabbccddeeff` is encoded as the bytes `33 22 11 00 55 44 77 66 88 99 aa bb cc dd ee ff`.

nishithvpoojary, 2021-03-12:
This is my HLApp after removing RTApp B and using only HLApp and RTApp. Here I have also removed the sending part from HLApp. 
------------------------------------------------------------------------------------------------------
#include <signal.h>
#include <string.h>
#include <stdio.h>
#include <ctype.h>
#include <stdbool.h>
#include <errno.h>
#include <unistd.h>

#include <sys/time.h>
#include <sys/socket.h>

#include <applibs/log.h>
#include <applibs/application.h>

#include "eventloop_timer_utilities.h"

/// <summary>
/// Exit codes for this application. These are used for the
/// application exit code. They must all be between zero and 255,
/// where zero is reserved for successful termination.
/// </summary>
typedef enum {
    ExitCode_Success = 0,
    ExitCode_TermHandler_SigTerm = 1,
    //ExitCode_TimerHandler_Consume = 2,
    //ExitCode_SendMsg_Send = 3,
    ExitCode_SocketHandler_Recv = 2,
    ExitCode_Init_EventLoop = 3,
    //ExitCode_Init_SendTimer = 6,
    ExitCode_Init_Connection = 4,
    ExitCode_Init_SetSockOpt = 5,
    ExitCode_Init_RegisterIo = 6,
    ExitCode_Main_EventLoopFail = 7
} ExitCode;

static int sockFd = -1;
//static int sockFd_B = -1;
static EventLoop* eventLoop = NULL;
//static EventLoopTimer* sendTimer = NULL;
static EventRegistration* socketEventReg = NULL;
static volatile sig_atomic_t exitCode = ExitCode_Success;

static const char rtAppComponentId[] = "CC0C53C7-E4B8-4D30-884C-034D6434DAAA";

static void TerminationHandler(int signalNumber);
//static void SendTimerEventHandler(EventLoopTimer* timer);
//static void SendMessageToRTApp(void);
static void SocketEventHandler(EventLoop* el, int fd, EventLoop_IoEvents events, void* context);
static ExitCode InitHandlers(void);
static void CloseHandlers(void);

/// <summary>
///     Signal handler for termination requests. This handler must be async-signal-safe.
/// </summary>
static void TerminationHandler(int signalNumber)
{
    // Don't use Log_Debug here, as it is not guaranteed to be async-signal-safe.
    exitCode = ExitCode_TermHandler_SigTerm;
}

/// <summary>
///     Handle send timer event by writing data to the real-time capable application.
/// </summary>
//static void SendTimerEventHandler(EventLoopTimer* timer)
//{
//    if (ConsumeEventLoopTimerEvent(timer) != 0) {
//        exitCode = ExitCode_TimerHandler_Consume;
//        return;
[... 133 lines cut ...]
    CloseFdAndPrintError(sockFd, "Socket");
}

int main(void)
{
    Log_Debug("High-level intercore application.\n");
    Log_Debug("Receives data from the real-time core.\n");

    exitCode = InitHandlers();

    while (exitCode == ExitCode_Success) {
        EventLoop_Run_Result result = EventLoop_Run(eventLoop, -1, true);
        // Continue if interrupted by signal, e.g. due to breakpoint being set.
        if (result == EventLoop_Run_Failed && errno != EINTR) {
            exitCode = ExitCode_Main_EventLoopFail;
        }
    }

    CloseHandlers();
    Log_Debug("Application exiting.\n");
    return exitCode;
}
-------------------------------------------------------------------------------------------------------------------

I use Starter Kit Rev2. My one more doubt was that if there were some problem with Component ID, there shouldn't have been any output displayed on the HLApp... Right?  And if HLApp can send data without encoding the ID, is it necessary to encode while sending from RTApp?

persello, 2021-03-12:
It's been really long since I last used Azure Sphere, and I don't use Windows/Visual Studio anymore, so I can't help much besides telling you that in the HL app you should put the RT app ID in a string, copy-pasted as you read it. In the RT app, you have to declare the HL app ID as an array. In this array, you have to use the byte order specified in my previous reply. 

No errors are printed because (I think) even without swapping the first bytes, you still have a valid ID, and you could have another app with that ID.

Please use triple backpacks when posting code.

nishithvpoojary, 2021-03-12:
Thank you very much for your response Persello.
I will try as you suggested.

johnnyhli, 2021-03-12:
@persello Thank you so much for your help! Hi @nishithvpoojary, if you still have issues, please let me know.

nishithvpoojary, 2021-03-15:
Yeah sure. Now I have some Azure Sphere SDK Fatal error. I will get back to it as soon as I solve this SDK problem.

tokenstar, 2021-07-05:
Since this issue has been idle for some months now, I am closing it.  Thanks @persello  and @nishithvpoojary  for your feedback and I hope the issues are resolved; please feel free to open a new issue as necessary.

closed 2021-07-05 by tokenstar