Azure/azure-sphere-samples #87: Trouble compiling large Cmake project in Visual Studio

k2nsloos, 2019-09-24:
Hi all,

I'm having trouble with my own CMake based Azure Sphere project using, using Visual Studio. The CMake File contains many source file entries. The compilation of the individual object files succeeds, but the final two commands fail. (Presumably the linking phase.)

[image]

Checking the build directory, I found these files present as expected:
[image]

But when comparing to smaller projects I noticed that no response file (.rsp) is generated in the smaller projects and compilation succeeds (for those smaller projects).
 
`  [110/113] cmd.exe /C "cd . && C:\PROGRA~2\MI3C21~1\Sysroots\2_BETA~1\tools\gcc\ARFCD8~1.EXE -B "C:/Program Files (x86)/Microsoft Azure Sphere SDK/Sysroots/2+Beta1905/tools/gcc" -march=armv7ve -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a7 --sysroot="C:/Program Files (x86)/Microsoft Azure Sphere SDK/Sysroots/2+Beta1905"  -nodefaultlibs -pie -Wl,--no-undefined -Wl,--gc-sections @CMakeFiles\BdrPlatform.rsp  -o BdrPlatform.out  && cd ."
  FAILED: BdrPlatform.out 
  cmd.exe /C "cd . && C:\PROGRA~2\MI3C21~1\Sysroots\2_BETA~1\tools\gcc\ARFCD8~1.EXE -B "C:/Program Files (x86)/Microsoft Azure Sphere SDK/Sysroots/2+Beta1905/tools/gcc" -march=armv7ve -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a7 --sysroot="C:/Program Files (x86)/Microsoft Azure Sphere SDK/Sysroots/2+Beta1905"  -nodefaultlibs -pie -Wl,--no-undefined -Wl,--gc-sections @CMakeFiles\BdrPlatform.rsp  -o BdrPlatform.out  && cd ."`

Any help, ideas and/or thoughts on the matter would be extreme welcome!

richardtaylorrt, 2019-09-25:
Hi - we note there's a missing "/" 
[image]
[image]

This may be an issue in your CMakeLists.txt (please check) or it could be a problem with our CMake toolchain.  If you suspect this, please could you help us narrow the problem down:
- Do you have any trouble running the existing samples in this repo?
- Is this an existing CMake project that works successfully outside of Azure Sphere?
- What do you see when [building via the command line outside of Visual Studio](https://docs.microsoft.com/en-us/azure-sphere/app-development/manual-build#compile-and-build-with-cmake)?
- Could you share a full paste of the verbose build output (copy/paste not just screenshot)?
- Are you able share your code/setup - e.g. is it already on a public GitHub repo somewhere?

If this is specific to running your code in Visual Studio, another option would also be to ask the [Visual Studio Developer Community](https://developercommunity.visualstudio.com/spaces/8/index.html).

k2nsloos, 2019-09-25:
Hi, 

First of all thanks for your reply and your time. 

Next regarding your comments:
- I started of by rerunning the SDK installer and doing a repair (just to be sure). 
- I tried to build the examples this works.
- If I add a small library with a few additional files this still works
- If I add a library with many additional files this fails.
- Using CMake using the commandLine yields the same error

I forked the example repository and modified the hello world example, in such a way that it fails compilation.  See: [https://github.com/k2nsloos/azure-sphere-samples/tree/setupNotWorkingExample/Samples/HelloWorld/HelloWorld_HighLevelApp_CMake](https://github.com/k2nsloos/azure-sphere-samples/tree/setupNotWorkingExample/Samples/HelloWorld/HelloWorld_HighLevelApp_CMake

Basically I added lots of source files under myLibrary/source/myfunctionXXX.c, each source file contains a single function, that does absolutely nothing.

For example myLibrary/source/myfunction123.c:
```
#include <myLibrary.h>

void myFunction123(void)
{

}
```

)

The regular CMakeLists.txt will fail compilation (at least on my machine), while the CMakeLists_works.txt does work. I hope this helps with narrowing it down...

jophippe, 2019-09-25:
Thank you for bringing this to our attention! This error is caused by CMake improperly handling response file generation by default when cross-compiling Windows to GCC, which we never noticed until now.
There is a workaround that we're implementing in our toolchain for an upcoming release. Follow these steps to try it out yourself:

- Close the project in VS and manually delete the .vs and out folders
- Navigate to your Azure Sphere SDK installation (usually C:\Program Files (x86)\Microsoft Azure Sphere SDK), then to the CMakeFiles subdirectory
- Create a backup copy of AzureSphereToolchainBase.cmake just in case, then open the file as administrator in a text editor
- Add this line anywhere in the file: `SET(CMAKE_COMPILER_IS_MINGW 1)`

After saving and closing the file, reopening the project in VS, and waiting for cache generation to finish, build should work as expected.
Hopefully this helps!

k2nsloos, 2019-09-26:
Yes!, that worked! Thank you very very very much!!!

closed 2019-10-03 by richardtaylorrt