Azure/azure-sphere-samples #88: Avnet Starter Kit wrong hardware define in sample_hardware?

jkuek, 2019-09-25:
Hardware\avnet_mt3620_sk\inc\hw\sample_hardware.h contains the following lines linking a LSM6DS3 to ISU1:

```
// MT3620 SK: Connect external LSM6DS3 to I2C using SOCKET1, pin MISO (SDA) and pin MOSI (SCL).
#define SAMPLE_LSM6DS3_I2C AVNET_MT3620_SK_ISU1_I2C

// MT3620 SK: Connect external LSM6DS3 to SPI using SOCKET1, pin MISO (MISO), pin SCK (SCLK), pin MOSI (MOSI) and SOCKET2 pin CS (CSB).
#define SAMPLE_LSM6DS3_SPI AVNET_MT3620_SK_ISU1_SPI

// MT3620 SPI Chip Select (CS) value "B". This is not a peripheral identifier, and so has no meaning in an app manifest.
#define SAMPLE_LSM6DS3_SPI_CS MT3620_SPI_CS_B
```

However the onboard LSM6DSO is connected to ISU2 not ISU1. See the block diagram here:
http://cloudconnectkits.org/sites/default/files/Block%20Diagram%20-%20Azure%20Sphere%20MT3620%20Starter%20Kit%20%28v2%29.pdf

richardtaylorrt, 2019-09-26:
Hi - there is a slightly confusing coincidence here:
- The Azure Sphere [I2C sample app](https://github.com/Azure/azure-sphere-samples/tree/master/Samples/I2C) happens to use the LSM6DS**3**.
- The Avnet Starter Kit (there are other Azure Sphere dev kits) happens to have an on-board LSM6DS**0**.
These two chips have different registers etc, so the DS3 sample app won't work against the onboard DS0.  There are two options:
1. Run the sample app unchanged against an external DS3, following the wiring instructions you found above.
2. Alter the sample app to run against the DS0 on ISU1.  You'd need to change the code and app_manifest.json accordingly.
Hope this helps.

jkuek, 2019-10-03:
Ah, got it. Thanks for the clarification.

closed 2019-10-03 by jkuek