Azure/azure-sphere-samples #90: Provide sample for accessing the onboard sensors

fmuntean, 2019-09-27:
Please provide sample accessing the on-board sensors for the Azure Sphere Starter Kit:
On-board sensors
3-Axis accelerometer
3-Axis gyro
Temperature
Pressure/Barometric

bawilless, 2019-09-30:
Avnet has provided an example application for the starter kit that exercises all the on-board sensors.  You can find the Blogs that detail how to find and use the examples here:  http://avnet.me/mt3620-kit-OOB-ref-design-blog

fmuntean, 2019-09-30:
Great,
Thank You!
It would be good to include the link above or the ling to the Avnet Git hub repository somewhere in the documentation, or the read.me page on this sample repository.

richardtaylorrt, 2019-10-01:
Hi - thanks. Yes, we will plan to update our docs and READMEs to help discovery of other samples for the different Azure Sphere dev kits that are available.

fmuntean, 2019-10-01:
Ok. Great.

closed 2019-10-01 by fmuntean