Azure/azure-sphere-samples #91: I2C app has wrong whoami value

fmuntean, 2019-09-27:
I2C accelerometer application starting.
INFO: WHO_AM_I=0x6c (I2CMaster_WriteThenRead)
ERROR: Unexpected WHO_AM_I value.
Closing file descriptors.
Application exiting.

Once I change the code to accept this value it works.

richardtaylorrt, 2019-10-01:
Hi - there is a slightly confusing coincidence here:
- The Azure Sphere [I2C sample app](https://github.com/Azure/azure-sphere-samples/tree/master/Samples/I2C) happens to use the LSM6DS**3**.
- The Avnet Starter Kit (note there are other Azure Sphere dev kits) happens to have an on-board LSM6DS**0**.

These two chips have different registers etc, so the DS3 sample app won't work against the onboard DS0.  There are two options:
1. Run the sample app unchanged against an external DS3, following the wiring instructions you found above.
2. Alter the sample app to run against the DS0 on ISU1.  As you've discovered, you'd need to change the code and app_manifest.json accordingly.
Hope this helps.

fmuntean, 2019-10-01:
Thank you.
would be good to have these instructions in the sample code so others do not run into the same issue.

closed 2019-10-01 by fmuntean