Azure/azure-sphere-samples #92: how to get a non blocking socket?

fmuntean, 2019-09-29:
tried the following:
`#include <stdio.h>
#include <sys/types.h>
#include <sys/socket.h>
#include <netdb.h>
#include <fcntl.h>

struct addrinfo hints, * res;
	int sockfd;

	// first, load up address structs with getaddrinfo():

	memset(&hints, 0, sizeof hints);
	hints.ai_family = AF_INET; //for IPv4 ; AF_UNSPEC for either IPv4 or IPv6;
	hints.ai_socktype = SOCK_STREAM; //TCP

	getaddrinfo(addr, port, &hints, &res);

	// make a socket:

	sockfd = socket(res->ai_family, res->ai_socktype, res->ai_protocol);
       fcntl(sockfd, F_SETFL, O_NONBLOCK);`
but it looks that F_SETFL is not defined ?

How do I get a non blocking socket ?

richardtaylorrt, 2019-10-01:
Hi - fcntl() is not present in the Azure Sphere SDK.  The SDK only includes APIs that Microsoft can presently support and maintain binary compatibility of across OS updates, and for the supported lifetimes of those devices.  The SDK intentionally excludes some standard POSIX APIs and Linux syscalls that do not presently meet these criteria, even if they are physically present within the kernel or OS.   

Instead, to get a non-blocking socket, it may help to look is this [file ](https://github.com/Azure/azure-sphere-samples/blob/master/Samples/PrivateNetworkServices/PrivateNetworkServices/echo_tcp_server.c) in one of our samples.

Specifically, the sending socket is opened as non-blocking:

    int sockType = SOCK_STREAM | SOCK_CLOEXEC | SOCK_NONBLOCK;
    serverState->listenFd = OpenIpV4Socket(ipAddr, port, sockType);

fmuntean, 2019-10-01:
That example is for getting a non-blocking socket bind to the local server port. I need to get a socket for client connection so don't need to bind(..) but connect(..) and tried already to change the bind to connect in the OpenIpV4Socket and is not working.

If there are some things intentionally excluded they should be documented as standard Linux api is to use the fcntl to make a non-blocking socket. 

I still need an example code that works to create a client non-blocking socket for communication with another server.

richardtaylorrt, 2019-10-01:
Hi - that's interesting.  Our dev teamm would expect this to be enough:

```
fd = socket(...,  <other options> | SOCK_NONBLOCK, ...);
connect(fd, ...);
```

In what way is this not working?

The dev team also advise that the connection may be async: i.e., connect() will return immediately. So you then have to wait (epoll/select/etc) on the socket fd and then when signalled check getsockopt() to see if the socket is writeable.

If this doesn't help, please share an example of the failing code.  Thanks.

fmuntean, 2019-10-04:
Thanks,
Tried it again and now is working as you explained above.

closed 2019-10-04 by fmuntean