Azure/azure-sphere-samples #94: Broken Pipe Exception and no PDB available

fmuntean, 2019-09-29:
ld-musl-armhf.so.pdb contains the debug information required to find the source for the module ld-musl-armhf.so.1

O keep getting broken pipe exception in visual Studio while using TCP socket and the PDB does not seems to be available. The stack trace also does not show any user code only : 	ld-musl-armhf.so.1![Unknown/Just-In-Time compiled code]

richardtaylorrt, 2019-10-01:
Hi - please could provide some more details of how to hit this problem.  Can you paste full output from Visual studio debug output?  Can you share example code that illustrates this?  Does it happen immediately, or only ocassionally or after some time?  Thanks.

fmuntean, 2019-10-01:
The way I got this problem was to get a client TCP socket connect to a server and try to receive data on it. As the socket was blocking and no data was received it timed out thus the error. As I mentioned there was nothing on the debug that I could get you other that what I mentioned above.
Can you point me to the pdb file for the above mentioned library? I could not find any pdb in the SDK folder.

richardtaylorrt, 2019-10-01:
Thanks for the extra information. We'll look into this further.

richardtaylorrt, 2019-10-03:
Hi - we don't currently supply debug symbols for the runtime libraries.

Is the socket timeout unexpected?  Did you expect different behavior from the runtime?  (If so, please tell us what you expected.)  Are you able to write signal handling code for this?

When the signal occurs, you should be able to go to the 'Threads' window and switch to the main thread to see your app's stack.

fmuntean, 2019-10-04:
Thanks for clarification on the debug symbols for the runtime libs.
I was able to get around the socket timeout.

closed 2019-10-04 by fmuntean