Azure/azure-sphere-samples #95: SIGKILL after 6hrs

fmuntean, 2019-09-30:
I am running the sample on I2C and so far it seems that the board stop working every 6 hrs:
INFO: 21653: vertical acceleration: -0.03g
INFO: 21653: Temperature: 29.12 C  84.42 F
INFO: 21654: vertical acceleration: -0.03g
INFO: 21654: Temperature: 29.21 C  84.58 F
INFO: 21655: vertical acceleration: -0.03g
INFO: 21655: Temperature: 29.14 C  84.45 F

Child terminated with signal = 0x9 (SIGKILL)

Ran the program twice and every single time after 21665 seconds (6:00hrs) I get the error above.

Can somebody please explain the behaviour?

I thought that the board supposed to keep running the app no matter what. and if the app is unresponsive was supposed to restart it.

richardtaylorrt, 2019-10-03:
Hi - unfortunately we couldn't repro this.  Are you running the published sample?  (I see there is "Temperature" output, which I don't think is part of the published sample.)   If you've made changes, please could you share these with us, so we can better advise.

Also, yes, in production, the OS would restart the app if it crashes.  However, this typically doesn't happen if you are running under a debugger.

fmuntean, 2019-10-04:
I was running the sample and added the temperature read from the sensor. I was also sending this data over the network to an MQTT server.
It was strange that SIGKILL happened at the exact same iteration.
I changed the iteration from int to unsigned int and so far I got no error but in the mean time I also moved to get the sensor data only once a minute too.

If you can't reproduce on your side then must be related to my project only. 
If happens again I will reopen this issue.

closed 2019-10-04 by fmuntean