Azure/azure-sphere-samples #96: invalid hardware defintion for the AVNET Starter Kit

fmuntean, 2019-10-01:
after changing the hardware definition for the project to the AVnet Starter Kit found that is invalid:
{"Name": "SAMPLE_LSM6DS3_I2C", "Type": "I2cMaster", "Mapping": "AVNET_MT3620_SK_ISU1_I2C", "Comment": "MT3620 SK: Connect external LSM6DS3 to I2C using SOCKET1, pin MISO (SDA) and pin MOSI (SCL)."}, 

but the I2C lsm6DS0 is actually wired to the other socket and changing it to:
{"Name": "SAMPLE_LSM6DS3_I2C", "Type": "I2cMaster", "Mapping": "AVNET_MT3620_SK_ISU2_I2C", "Comment": "MT3620 SK: Connect external LSM6DS3 to I2C using SOCKET1, pin MISO (SDA) and pin MOSI (SCL)."},

fixed it.

Maybe the LSM6DS3 should also be renamed to the real LSM6DS0 included with the board too.

richardtaylorrt, 2019-10-03:
Hi - as per my answer to [this issue](https://github.com/Azure/azure-sphere-samples/issues/91), please note there are many dev kits for Azure Sphere, and there is a slightly confusing coincidence with the Avnet Starter kit:

- The Azure Sphere [I2C sample app](https://github.com/Azure/azure-sphere-samples/tree/master/Samples/I2C) happens to use the LSM6DS**3**, the idea being that you wire this part to your board.
- The Avnet Starter Kit happens to have an on-board LSM6DS**0**.

These two chips have different registers etc, so the DS3 sample app won't work against the onboard DS0.  There are two options:
1. Run the sample app unchanged against an external DS3, following the wiring instructions in the "Comment" field in the hardware definition.
2. Alter the sample app to run against the DS0 on ISU2.  As you've discovered, you'd need to change the code and app_manifest.json accordingly (including the registers used, and the hardware definition constants.)

It looks like you've taken option 2, which is great if it is working for you.

We will add documentation to our samples README to help explain this.

fmuntean, 2019-10-04:
Then which of the hardware targets in this repository match the AVNET Starter KIT board?
I thought that by changing the **Target Hardware Definition** Folder to ..\Hardware\avnet_mt3620_sk will use this board definition thus it should match the hardware provided with the board.

Isn't this the reason to use the Target Hardware Definition inside Visual Studio ?

richardtaylorrt, 2019-10-06:
Yes, that's the correct Hardware Definition for the Avnet starter kit.

You are correct that perhaps this definition should also define constants for the DS0 and other built-in parts on the Starter Kit.  E.g. we should probably define AVNET_MT3620_SK_ACCELEROMETER_LSM6DS0 (mapping to AVNET_AESMS_ISU2_I2C).  I'll raise a task to track this.

But even if we added this definition, it wouldn't automatically allow the existing sample app code for the DS3 part to work for a different part (the DS0).   To use this code unchanged, you'd need to find and wire up a DS3 part using the "Comment" instructions (as per option #1 above).

As I say, I appreciate the DS3 vs DS0 co-incidence is confusing, and we'll add further documentation to explain.

fmuntean, 2019-10-07:
I understand the issue with the sample and confusion on the parts.
This ticket is about having the correct hardware definition for the board also  having defined a board ID in these hardware definition files could help with the sample accommodating to the board and part itself. 
The WhoAmI() method could use the response to know what part is used instead of blocking the code.
Also could be better if instead of prefixing all those hardware definitions with the board ID maybe just define the functional part instead like ACCELEROMETER_I2C  instead of AVNET_MT3620_SK_ACCELEROMETER_LSM6DS0 as this could help with running the same sample on multiple boards just by replacing the hardware definition and would allow to run them on custom hardware too.

richardtaylorrt, 2019-10-14:
Thanks.

We will track the idea of making the sample code work for both the DS3 and DS0.  As you suggest, we would need to:

- test the WhoAmI response
- change SAMPLE_LSM6DS3_I2C to be SAMPLE_ACCELEROMETER_LSM6DSX_I2C or similar.  (We wouldn’t want to pretend the sample would magically work for any accelerometer!)

ld0614, 2019-11-21:
I just ran into this same issue, the only code changes I needed to get the sample booting were:

- Update the device ID to AVNET_MT3620_SK_ISU2_I2C
https://github.com/Azure/azure-sphere-samples/blob/d595f6bc697ec84f55d49b81908bb444f3af1ba0/Samples/I2C/I2C_LSM6DS3_HighLevelApp/main.c#L252
- Change the WhoAmI address to 0x6C as per this [datasheet](https://www.st.com/resource/en/datasheet/LSM6DSO.pdf)
https://github.com/Azure/azure-sphere-samples/blob/d595f6bc697ec84f55d49b81908bb444f3af1ba0/Samples/I2C/I2C_LSM6DS3_HighLevelApp/main.c#L117

Having this sample has been invaluable as I've not used I2C before and having a prebuilt example to hack into a working program has removed a massive barrier to entry.

Jeff-Schafer, 2020-01-10:
Thanks, @ld0614.
We are tracking an update to the docs.

Closing this issue now.

closed 2020-01-15 by Jeff-Schafer-Azure