Azure/azure-sphere-samples #100: Small edit for "Run the sample with an IoT hub"

djaus2, 2019-10-14:
Under the heading: "Configure the sample application to work …"
Step 6.  I suggest you add an example of what is required 
eg Append the following: to step 6:
For example:
`  "Capabilities": {
    "AllowedConnections": [ "global.azure-devices-provisioning.net", "mysphere.azure-devices.net" ],`
Where mysphere.azure-devices.net  is your the hostname for the Azure IoT Hub.

djaus2, 2019-10-14:
Also under "Build and run the sample"
Can I suggest the use of **Device Explorer** be mentioned as an alternative.
I had problems with login for **Cloud Explorer**, and I am sure others probably do.
[Device Explorer on GitHub](https://github.com/Azure/azure-iot-sdk-csharp/tree/master/tools/DeviceExplorer)
[Download Device Explorer Installer](https://github.com/Azure/azure-iot-sdk-csharp/releases/download/2019-1-4/SetupDeviceExplorer.msi)

Jeff-Schafer-Azure, 2020-01-15:
On the [Device Explorer on GitHub](https://github.com/Azure/azure-iot-sdk-csharp/tree/master/tools/DeviceExplorer) page note that it says:

**Device Explorer is deprecated as of 10/30/2019 and is being replaced by [Azure IOT Explorer](https://github.com/Azure/azure-iot-explorer) . [This](https://github.com/Azure/azure-iot-sdk-csharp/releases/tag/2019-9-11) is the last release of Device Explorer.**

closed 2020-01-15 by Jeff-Schafer-Azure