Azure/azure-sphere-samples #102: Mutual authentication

dkarakas, 2019-10-21:
I am currently trying to send the x509 certificate(to my understand produced as a result of remote attestation) to an API so that Azure Sphere can authenticate itself. I have been follow the explanation here: https://docs.microsoft.com/en-us/azure-sphere/app-development/curl

After using MyCallback, I get the following output:
Press button A to initialize a set of parallel, asynchronous web transfers.
Opening SAMPLE_LED
Opening SAMPLE_BUTTON_1 as input.
Using libcurl/7.61.0 wolfSSL/4.1.0 c-ares/1.14.0

*   Trying ***...
* TCP_NODELAY set

* Connected to *** port 443 (#0)
* successfully set certificate verify locations:
*   CAfile: ***
  CApath: /etc/certs/
ssl func callback error 2
* ALPN, offering http/1.1

* ALPN, server did not agree to a protocol
* SSL connection using TLSv1.2 / ECDHE-RSA-AES256-GCM-SHA384
> GET *** HTTP/1.1

(*** are masked output values)

I am not sure how to debug "ssl func callback error 2." 
It seems that "Loading the device auth certificate for this tenant failed."

Any suggestions how to debug this?

richardtaylorrt, 2019-10-21:
Hi - you may also need to add/update the `DeviceAuthentication` field in the app_manifest.json of your app.  Its value should be set to your tenant ID (get this from: azsphere tenant show-selected).

If this is the problem, then apologies, we'll make sure to update the docs to cover this point.

If it isn't the problem, let us know, and we'll investigate further with you.  Thanks.

dkarakas, 2019-10-22:
Hi,
Thank you! Updating DeviceAuthentication field fixed the issue! 

General question: Is the client certificate part of the header or the body once requested by the API? I am having trouble making it work with Azure Function. (I will close the issue once I receive an answer)

richardtaylorrt, 2019-10-22:
Great to hear the issue is fixed.

I'll ask the team if they have any further advice here.  One possibility is that some PaaS services only support mutual authentication via "secure renegotiation" of the HTTPS connection but that libcurl (the HTTPS client library we're using here) does not easily support this. I'm not sure if this is true of Azure Functions.

dkarakas, 2019-10-22:
I see. Maybe in the coming days I will set up a local API and see if I can make a Client Authentication work in the initial handshake(as appose to "secure renegotiation"). Please, do let me know if they have any further advice. It will be much appreciated

dkarakas, 2019-11-29:
I was able to get the mutual authentication working and if I have time I will share the steps I took, which might be helpful to other people. Just one last feedback: please add a link to the root certificate("Microsoft Azure Sphere Root") in https://docs.microsoft.com/en-us/azure-sphere/app-development/curl, so that the client's certificate chain can be verified(using tenant's certificate is problematic). I am closing the issue, since it has been resolved.

closed 2019-11-29 by dkarakas