Azure/azure-sphere-samples #105: not doing download because here is no internet connectivity

floresboy, 2019-11-13:
Output is repeated 
not doing download because here is no internet connectivity while 

**azsphere device wifi show- status** show as state 'connected' and all relevant ports are open on the Wifi network. and suggestion what might be wrong. read some stuff on timing and NTP server on Sphere OS 19.10
thanks K

richardtaylorrt, 2019-11-14:
Hi @floresboy - please could you provide a bit more detail.  Are you running one of our sample apps?  (E.g. are you running the HTTPS sample app?)  What is it that is not downloading?  Please could you paste the output from the device.

Azure Sphere has the following [networking requirements](https://docs.microsoft.com/en-us/azure-sphere/network/ports-protocols-domains).  Please can you confirm your device is attached to a network that meets these requirements.  Thanks.

floresboy, 2019-11-14:
Sorry i should mention that explicitly; this is while running the unchanged CURLeasy sample (https://github.com/Azure/azure-sphere-samples/tree/master/Samples/HTTPS/HTTPS_Curl_Easy)

 yes, all the req are met and tested on a port level and the azsphere device wifi show- status show as state 'connected'. all other wifi connected device work perfectly. Some assistance is highly appreciated so thanks in advance. K

richardtaylorrt, 2019-11-14:
Hi - please can you:

- copy the output you see from the device (in Visual Studio "Device Output" pane).
- run "azsphere get-support-data -o data.zip"
- send this output and the zip file to AzCommunity@microsoft.com (do not attach it here), with a link back to this issue.

Thanks.

floresboy, 2019-11-14:
done that and send it to mailto:AzCommunity@microsoft.com

floresboy, 2019-11-19:
tested some more but nothing yet. Just the plain curl example and a working wifi connection, no connection. Anyone idea's ?

richardtaylorrt, 2019-11-19:
Hi @floresboy - the team are investigating this.  In the meantime:

- Does the same problem happen if you connect the device to another wifi network (eg. a hotspot on your cellphone)?
- You could also ask for advice on our [MSDN forum](https://social.msdn.microsoft.com/Forums/en-US/home?forum=azuresphere).

floresboy, 2019-11-19:
> Hi @floresboy <https://github.com/floresboy> - the team are investigating this. In the meantime:

> Does the same problem happen if you connect the device to another wifi network (eg. a hotspot on your cellphone)?

tried that (even with the neighbors wifi ;-)
> You could also ask for advice on our MSDN forum <https://social.msdn.microsoft.com/Forums/en-US/home?forum=azuresphere>

good suggestion, Thanks a lot.

firedog1024, 2019-11-27:
@floresboy  I had the same issue as you with thissample, it was working fine for a while then quit I think when I changed the wifi I was using (work -> home).  Tried a bunch of stuff including phone as a wifi hotspot and nothing allowed me to connect to the internet even though the device was connected fine to the wifi.  I was able to resolve this by doing a recover on the device and basically hard resetting it back to a clean image.  

From the Azure Sphere Command line run:
`azsphere device recover`
`azsphere device enable-development`
`azsphere device wifi add -s <SSID> -p <password>`

Hopefully that gets the device to connect using the sample, it did for me.

floresboy, 2019-11-27:
Hi firedog1024, Thanks a lot for sharing this info. i'll give it a go and let you know. Once again, Thanks! K

RobertoD91, 2019-11-29:
Same issue, device restore don't help

but if I comment the `goto exitLabel;` around line 100, seems to work

```
bool isNetworkingReady = false;
if ((Networking_IsNetworkingReady(&isNetworkingReady) < 0) || !isNetworkingReady) { 
   Log_Debug("\nNot doing download because there is no internet connectivity.\n");
   Log_Debug("%s (%d).\n", strerror(errno), errno);
   //goto exitLabel;
}
```

maybe the problem is `Networking_IsNetworkingReady` ?

firedog1024, 2019-11-30:
@RobertoD91 I think you and I may have different issues because I did the same thing in my code by basically bypassing the `Networking_IsNetworkingReady` check and the sample failed to download the example.com web page contents.  Sorry my fix did not resolve the issue, I'm interested to see if this fixes the issue for @floresboy.  If not I guess I'm the odd man out.

floresboy, 2019-11-30:
Hi @RobertoD91 and firedog1024 

Thanks for your comments and help. 

Even after the suggested  'azsphere device recover ....etc ' and an apparently  good Wifi connection from 'azsphere wifi status', no succes on the unmodified sample code for HTPS_Curl_Easy. So i modified as suggested by RobertoD91 along with HTTP instead of HTTPS (line 122) to grab the example.com page. That worked! 

So i agree maybe the problem is Networking_IsNetworkingReady ?

floresboy, 2019-11-30:
and to add to tat i had also remove the InterFaceState check in line 260. If left in place it generated an error 5. puzzling and not very convincing for now

floresboy, 2019-12-16:
Any news on this one ? I can get it working when I remove theif ((Networking_IsNetworkingReady(&isNetworkingReady) < 0) || !isNetworkingReady) check and refer to a non ssl (http) website. So two things now working in this example (wifi check and ssl). Can you fix or some guidance ? Thanks Kamiel

kangsu, 2019-12-20:
For the expression in the if-statement, do you know what is returning false?

if ((Networking_IsNetworkingReady(&isNetworkingReady) < 0) || !isNetworkingReady)

If the Networking_IsNetworkingReady call is returning a value less than 0, then can we get the errno to try to help diagnose?

floresboy, 2019-12-20:
will check in the upcoming days

floresboy, 2019-12-23:
will check the error value later.
Networking_IsNetworkingReady: Verifies whether internet connectivity is available and t**ime is synced.**

i have the suspicion that Time (of RTC) is not correct and that this is causing the failure on the check. This because networking IS working actually. Any idea how th check this ?

floresboy, 2019-12-24:
```
if (Networking_IsNetworkingReady(&isNetworkingReady) < 0) {
       Log_Debug("\nError getting network status, IsNetworkinReady returned -1 error: isNetworkingReady=%d.\n", isNetworkingReady);
   }
   else {
       Log_Debug("\nGetting networkstatus in boolean went OK (non -1) with: isNetworkingReady=%d.\n", isNetworkingReady);
   }

   if (isNetworkingReady)
   {
       Log_Debug("\nThe network is reported as Ready\n");
   }
   else
   {
       Log_Debug("\nThe network is reported as NOT ready\n");
       // goto exitLabel;
   }

```
resulted in 

cURL easy interface based application starting.

This sample periodically attempts to download a webpage, using curl's 'easy' API.
Setting networkstatus in boolean went OK with: isNetworkingReady=0.

The network is reported as NOT ready

 -===- Starting download -===-
curl_easy_setopt CURLOPT_URL (curl err=0, 'No error')

> so, it seems isNetworkingReady is returning a 0 (false) while networking is OK proved by the correct http download. (time issues?)

spatdsg, 2020-02-18:
@floresboy  or @RobertoD91  - are you still having this issue? Best we can tell at this point the devices are connected to the internet , but time sync is not working. I think someone had tested on a PC on the same network and ran some W32time commands and it worked from a PC on the same network ( proving the ports weren't blocked).  If you are still having issues, I'd like to try to understand where it may be failing,

floresboy, 2020-02-19:
didn't test further cause of other prior's but its still an issue.

floresboy, 2020-05-14:
retested but still the same:
a) a need to remove the check cause false 'no Internet' detection
b) httpS not working >< httP is.
have no clue. Help is appreciated

kangsu, 2020-05-14:
Which version of the OS is the device running?  For example, 20.04?

floresboy, 2020-05-14:
20.04, but was issue also months ago with 19.x. When I remove Niet check and SSL the Avnet board board works OK. I can’t see why de sample doen not work.

johnnyhli, 2020-10-20:
Closed, since It appears the issue is not reproducible after updating to 20.07.

closed 2020-10-20 by johnnyhli