Azure/azure-sphere-samples #109: curl_easy_cleanup hangs application HTTPS_Curl_Multi

ld0614, 2019-11-21:
I create a project using the default sample code for HTTPS_Curl_Multi and modifying the colour of the LED flashing.

As I'm new to Sphere I hadn't worked out I needed to declare the change in the app manifest.  This caused the application to fail as it didn't have the permissions to open the different colour LED.

What was more confusing was that rather than completely failing, Visual Studio continued to act like the application was running.  After tracing through with the debugger it appears to hang at https://github.com/Azure/azure-sphere-samples/blob/d595f6bc697ec84f55d49b81908bb444f3af1ba0/Samples/HTTPS/HTTPS_Curl_Multi/web_client.c#L434

This stops the UI cleanup and the application to never actually close, requiring a manual stop in Visual Studio.

richardtaylorrt, 2019-11-26:
Hi @ld0614 - thanks for raising this.  We agree the app cleanup code should cope with this better, and we'll fix in an upcoming release.

johnnyhli, 2020-02-25:
Hi @ld0614 – We recently have new update for the Sample apps. This should be fixed now.

closed 2020-02-27 by johnnyhli