Azure/azure-sphere-samples #115: IoT Central Application instructions need to be updated (ShowIoTCentralConfig.exe) no longer supported

bawilless, 2019-12-24:
The IoT Central instructions for this example refer to the ShowIoTCentralConfig.exe utility to pull the application details required to connect the Azure Sphere Device to the IoT Central Application.  This utility is no longer supported (or so I was told) so it can't be used to get the required details.  Here's the output when I try to use the utility:

PS C:\Users\051520\source\repos\azure-sphere-samples-noModifications\Samples\AzureIoT\Tools> .\ShowIoTCentralConfig.exe
Tool to show Azure IoT Central configuration for Azure Sphere applications

Are you using a Work/School account to sign into your IoT Central Application (Y/N) ?n
Enter the Scope Id from the IoT Central App 'Administration | Device Connection' >0ne000A7***
Enter the Primary Saas Key from the IoT Central App 'Administration | Device Connection' >G61ePJbKoAw2ENAMUfAQlNYnIkK2pSt/1Cr5********
Enter any real Device Id from the IoT Central App >d53f0069-6d28-471c-beb0-562810d0b77e
Could not get the IoT Hub endpoint information.
PS C:\Users\******\source\repos\azure-sphere-samples-noModifications\Samples\AzureIoT\Tools>

kkomand, 2020-01-08:
The IoT Central interface has changed.   Need  IoT Central Application instructions  to be updated. As well the ShowIoTCentralConfig.exe is not working

ozhernan, 2020-01-08:
Looks like ShowIoTCentralConfig.exe doesn't list the new IoT central applications, it only list the ones created using the Legacy application(2018) template from Azure Portal.

Jeff-Schafer, 2020-01-10:
Hi ozhernan:
You are correct, "ShowIoTCentralConfig.exe" only works with "Legacy application (2018) template" ("regular non-preview IoT Central applications").

We are looking into how to update this tool, and our instructions, to support preview IoT Central applications.  In the meantime, you may be able to use [this tool](https://github.com/Azure/IoTC-Reserved-IoTHub-List) to get the IoT Hub URLs you need.

lwoo-code, 2020-01-31:
Hi, could you update the IoT Central example instruction? The following process will work for the new IoT Central template.

The current issue seems to only affect if you select the "Y" in "Are you using a Work/School account to sign into your IoT Central Application (Y/N) ?"

We can still manually fill in the rest and get the endpoint of IoT Hub:
1)	When the application prompt if you are using Work/School account, response “N” to go for manual setting route
2)	Enter the scope ID for your DPS that you can find under IoT Central -> Administration -> Device Connection -> ID Scope
3)	For Primary SaaS key, enable the options of “SAS tokens on this app” -> View Keys -> Copy the primary key
4) Manually create a device with your Sphere device ID in IoT central, and copy and paste your device ID in LOWER case into the tool.

Appreciate if you could update the above steps to our AzureIoT example as workaround for the time being while we are waiting for the long term fix on the tool.

johnnyhli, 2020-02-25:
Hi @bawilless -  Thanks for your patience. We recently have new update for the Sample apps. This should be fixed now.

closed 2020-02-27 by johnnyhli