Azure/azure-sphere-samples #116: ShowIoTCentralConfig.exe sometimes doesn't allows to select the IoT central application

ozhernan, 2020-01-08:
When more than one IoT central applications are listed in the tool, for example:

_Are you using a Work/School account to sign into your IoT Central Application (Y/N) ?y
Getting your IoT Central applications
IoT Central applications:
1: TTT-Redmond
2: IoT Portal
**Select (1-2) >**_
 
Around 50% of the times, the application wont stop in the prompt in bold above to type the number but instead terminates immediately.

Jeff-Schafer, 2020-01-10:
Hi ozhernan:
Thank you for bringing this to our attention.

Currently, we are looking in to it.

We'll post a reply when we have a solution for this.

Jeff-Schafer, 2020-01-13:
Hi ozhernan:
We've found that typing **Y** & **Enter** immediately, the **Enter** key is buffered. When the **Select** prompt is displayed, the **Enter** key is processed & is considered a "no selection", so the app exits.

A workaround would be to just type in **Y** without pressing **Enter**.
The **Select** prompt should be displayed & waiting for user input.

Is this what is occurring in your IoT Central app?

ozhernan, 2020-01-13:
yes, that is the issue, just typing Y is the workaround. Thanks.

Jeff-Schafer, 2020-01-14:
Hi ozhernan:
Thanks for confirming this is the issue.

closed 2020-01-14 by Jeff-Schafer-Azure