Azure/azure-sphere-samples #117: How to get the Device ID and Device Certificate from application, not from command line.

v-jiacz, 2020-01-09:
I need the Device ID and Device Certificate when porting azure-iot-sdk-c(public-preview) to the board.

Jeff-Schafer, 2020-01-09:
Hi v-jiacz:

In the Azure Sphere Developer Command Prompt, run the following commands:

Device ID:
     **azsphere device show-attached**

To get the Device Certificate, run:
     **azsphere tenant download-CA-certificate --output CAcertificate.cer**

It will need to be validated. Run:
     **azsphere tenant download-validation-certificate --output ValidationCertification.cer --verificationcode** <verification code>

Jeff-Schafer, 2020-01-09:
OK. You want to get this via the applibs APIs.
Sorry, misunderstood, will get back to you.

Jeff-Schafer, 2020-01-10:
Hi v-jiacz:
Please can you tell us more about what you are trying to achieve here.
There is an integrated version of the IoT C SDK already built in to the Azure Sphere device software.

But perhaps you are looking to use a newer version?  If so, which version, and which specific feature of it are you trying to use?

Currently, there is no Azure Sphere API for an app to separately obtaining the device ID or device certificate.  I haven’t investigated in detail, but this may mean it is not possible to port a new/different version of the IoT C SDK to Azure Sphere.

v-jiacz, 2020-02-28:
The libazureiot.so library under /opt/azurespheresdk/Sysroots/3/usr/lib/libazureiot.so folder only support Azure IoT Central Application(legacy).
Could you rebuild a [azure-iot-c-sdk(public-preview)](https://github.com/Azure/azure-iot-sdk-c/tree/public-preview) version. In that case, we can connect to the Azure IoT Central(public-preview), and use the new features to develop our device application.

closed 2020-02-28 by v-jiacz