Azure/azure-sphere-samples #119: Can't connect internet by Ethernet on ect mode

Shaing, 2020-01-15:
Here is my way: 
Sideload Sphere APP(every setting is fine including device json config.) and using ethernet on edv mode, it work well. But when I enable ect mode, it can't connect to internet on ect mode.

* edv: enable-development (by azsphere)
* ect: enable-cloud-test (by azsphere)

Please help, if I am missing something.

Thx.

bawilless, 2020-01-15:
When you put the device into ect mode, all applications are removed from the device.  You have to setup the over the air updates to define what application(s) get pushed to the devices.  If you're using the wired ethernet, make sure you also include the board configuration package enc28j60-isu0-int5.imagepackage in your OTA update configuration.

You may have the issue where you need to use the device with ethernet but since the board configuration package is not on the device, it can't connect to the internet to actually get the board configuration package.  If this is the case, one work around would be to configure the device on the wifi network to get the initial OTA update, that will include the board configuration package.  After that you should be good.

richardtaylorrt, 2020-01-15:
Thanks for your question @Shaing and for your response @bawilless .  See also the documentation on [Cloud deploy a board configuration image package](https://docs.microsoft.com/en-us/azure-sphere/network/connect-ethernet#cloud-deploy-a-board-configuration-image-package).

Shaing, 2020-01-17:
Thank you, guys! But problems still exist...
I has been confirmed the enc28j60-isu0-int5.imagepackage by "azsphere dev img list-installed" on ect mode.

@richardtaylorrt Can I download the "enc28j60-isu0-int5.imagepackage" by OTA? I has tried but it didn't work. 

Now, I checked:
1. json setting is OK
2. img is OK
3. wifi setting is OK
but when I switch to ect mode, I got this error code (I used UART to get the log)
> ERROR: Networking_SetInterfaceState for interface 'eth0' failed: errno=2 (No such file or directory)

and I switch edv at the same time, it works fine. Same App version, just switches mode and got different result.

bawilless, 2020-01-17:
I think the issue is that the enc28j60-isu0-int5.imagepackage package is not in your OTA deployment since your runtime debug can't open eth0.  

I've got a deployment that uses the ethernet interface and I remember adding the enc28j60-isu0-int5.imagepackage file to the OTA deployment.  I don't have a record of this action, but I remember doing it.  What's the error you get when you try to add this package to your deployment?  

Also, it does not make sense that you can still see the enc28j60-isu0-int5.imagepackage on your device when you go into ect mode.  Entering that mode should remove all applications on your device.  I would try to recover your device "azsphere device recover" to erase the devcie and put it back into a factory state.  Note that this operation will also remove any wifi networks you have configured on the device.

Shaing, 2020-01-22:
@bawilless Big thx for your support, and I'd solved this problem !!!
I found these rules:
1. Create a product group and there were 4 default device groups.
2. This issue is that I also create our device group but customer dg only supports one img-package which means App img or Board cfg img just one type can exists in the customer dg.
3. but when I use "Production"  dg of product group, it works finally !!!
4. Using "azsphere dev img list-targered" to check all img on the cloud that all can using by OTA. And I can see the both App img and Board cfg.

Summary:
azsphere dev ect -dgn "Production" -pn [pn] to enable OTA mode with dg: "Production" and pn "[pn]".

You guys suggestion is good. Indded, On ect mode, if Board cfg img not in cloud and device will can't connect to network.

closed 2020-01-22 by Shaing

Shaing, 2020-02-05:
Hi guys, 
if you are referencing this issue, I need to update the result.
the issue is no matter what device group if we need to see both app and board config in the list-targeted, you need to upload in the same time, for example:
> azsphere device-group deployment create --devicegroupname Production --productname dfiiotdemo --filepath=enc28j60-isu0-int5.imagepackage,AzureIoT.imagepackage -f  

Buz the cmd will overwrite the cloud img on the azure security service. if you only upload the one of both, it always be overwrited by each other.

yes, the result as the doc metioned. 
https://docs.microsoft.com/zh-tw/azure-sphere/deployment/create-a-deployment#create-a-deployment-for-a-device-group

pospospos2007, 2022-05-11:
I got the same error:
ERROR: Networking_SetInterfaceState for interface 'eth0' failed: 
I still don't get your point.

pospospos2007, 2022-05-11:
Is that device OS version issue or something else?