Azure/azure-sphere-samples #120: AZURE_SPHERE_PROV_RESULT_PROV_DEVICE_ERROR with Azure IoT Sample

LennardSchmit, 2020-01-17:
Hi,

I am using the Azure IoT sample for my application, now since the 19.11 update i mostly get the following error when connecting to the IoT Hub. 
AZURE_SPHERE_PROV_RESULT_PROV_DEVICE_ERROR
This error also occurs when using the Azure IoT Sample.
My SDK and devices are using 19.11, and my IoT Hub and DPS are both S1 tier.

Shaing, 2020-01-22:
Check your json file(app_manifest.json) and need to add your URL of IoT hub and DPS(Scope Id).

For example
```
{
  "SchemaVersion": 1,
  "Name": "sample",
  "ComponentId": "819255ff-8640-41fd-aea7-f85d34c491d5",
  "EntryPoint": "/bin/app",
  "CmdArgs": [ "0ne000596BA" ], // Scope id of DPS
    "Capabilities": {
        "AllowedConnections": [ "global.azure-devices-provisioning.net", "[URL of IoT hub]" ],
      "Gpio": [ "$MT3620_RDB_BUTTON_A", "$MT3620_RDB_BUTTON_B", "$MT3620_RDB_LED1_RED", "$MT3620_RDB_HEADER1_PIN4_GPIO", "$MT3620_RDB_HEADER1_PIN5_GPIO" ],
      "Uart": [ "$MT3620_RDB_HEADER4_ISU1_UART"],
        "DeviceAuthentication": "[Tanent Id]",
        "NetworkConfig": true
    },
  "ApplicationType": "Default"
}
```

LennardSchmit, 2020-01-22:
This issue has since been resolved, now it connects to the DPS correctly. It didn't require changes in my code. Perhaps it was a bug in the Azure Sphere Security Service, but i just had authentication problems for a few days.

closed 2020-01-22 by LennardSchmit