Azure/azure-sphere-samples #125: Build/Deploy/Debug doesn't work on CLI after update

galvesribeiro, 2020-02-24:
Hello folks!

After the latest update I'm getting this error while trying to sideload and run or debug the apps from Linux CLI:

```
azsphere device sideload deploy --manualstart --imagepackage out/debug/backbone.imagepackage
Deploying 'out/debug/backbone.imagepackage' to the attached device.
warn: This app targets Beta APIs that may change or be removed in future.
Image package 'out/debug/backbone.imagepackage' has been deployed to the attached device.
The application was not automatically started. Use 'azsphere device app start' to start it.
```

Without debug:
```
azsphere device app start --componentid 0f149356-2cb8-4aa4-9608-f41e060d82c5
error: Could not start component '0f149356-2cb8-4aa4-9608-f41e060d82c5': Device error 1.15.
```

With debug:

```
telnet -a 192.168.35.2 2342
Trying 192.168.35.2...
Connected to 192.168.35.2.
Escape character is '^]'.
Starting debugger....
gdbserver: Cannot exec /mnt/apps/0f149356-2cb8-4aa4-9608-f41e060d82c5/bin/app
gdbserver: Error: No such file or directory

During startup program exited with code 127.
Exiting
Connection closed by foreign host.

```

Am I missing something?

Thanks!

johnnyhli, 2020-02-25:
Hi @galvesribeiro - Thank you for bringing this. Can you run` azsphere device app show-status` and attach the results? Thanks.

galvesribeiro, 2020-02-25:
Hello @johnnyhli thanks for getting back. Sorry I forgot I had this issue open and didn't report back. I found the reason.

The error was because with the latest SDK the linker commands wasn't setting the proper interpreter flag when building the binary from the CLI. I've added `-Wl,-dynamic-linker,/lib/ld-musl-armhf.so.1` to the linker and it is working now. Without that, it was setting `/lib/ld-musl-arm.so.1` as the interpreter, hence why the `No such file or directory` error. 

The application start and I can manually attach `gdb` (from the SDK directory) on it. However I have an issue trying to use VSCode and `cppdbg` with it to debug.

The debugger only starts after the telnet is connected, so the `gdbserver` port 2345 is only available after that, and only while the telnet session is up. That brings problems with VSCode and `cppdbg`.

Here is my configuration:

`tasks.json`:

```json
{
    "version": "2.0.0",
    "runner": "terminal",
    "options": {
        "cwd": "${workspaceRoot}/Firmware",
        "env": {
            "PATH": "${env:PATH}:${workspaceFolder}/../build-tools/bin/linux"
        }
    },
    "tasks": [
        {
            "label": "4-attach",
            "type": "shell",
            "presentation": {
                "echo": true,
                "reveal": "always",
                "focus": false,
                "panel": "new",
                "showReuseMessage": true,
                "clear": false
            },
            "isBackground": true,
            "dependsOn": "3-start-debugger",
            "command": "telnet",
            "args": [
                "192.168.35.2",
                "2342"
            ]
        },
        {
            "label": "3-start-debugger",
            "type": "shell",
            "dependsOn": "2-deploy-image",
            "command": "azsphere device app start --debug --componentid 0f149356-2cb8-4aa4-9608-f41e060d82c5",
            "presentation": {
                "echo": true,
                "reveal": "always"
            },
            "group": {
                "kind": "build",
                "isDefault": true
            }
        },
        {
            "label": "2-deploy-image",
            "type": "shell",
            "dependsOn": "1-build-firmware",
            "command": "azsphere device sideload deploy --manualstart --imagepackage out/debug/eGYM_Backbone.imagepackage",
            "presentation": {
                "echo": true,
                "reveal": "always"
            }
        },
        {
            "label": "1-build-firmware",
            "type": "shell",
            "command": "./rungn debug -p",
            "presentation": {
                "echo": true,
                "reveal": "always"
            },
            "problemMatcher": [
                {
                    "owner": "cpp",
                    "fileLocation": [
                        "relative",
                        "${workspaceRoot}/Firmware"
                    ],
                    "pattern": {
                        "regexp": "^../../(.*):(\\d+):(\\d+):\\s+(warning|\\w*\\s?error):\\s+(.*)$",
                        "file": 1,
                        "line": 2,
                        "column": 3,
                        "severity": 4,
                        "message": 5
                    }
                },
                {
                    "owner": "cpp",
                    "fileLocation": [
                        "relative",
                        "${workspaceRoot}/Firmware"
                    ],
                    "pattern": {
                        "regexp": "^../../(.*?):(.*):\\s+(warning|\\w*\\s?error):\\s+(.*)$",
                        "file": 1,
                        "severity": 3,
                        "message": 4
                    }
                }
            ]
        },
        {
            "label": "kill",
            "type": "shell",
            "command": "killall -SIGKILL telnet",
            "presentation": {
                "echo": false,
                "reveal": "never",
                "focus": false,
                "panel": "shared",
                "showReuseMessage": false,
                "clear": false
            }
        }
    ]
}
```

`launch.json`:

```json
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "eGYM.Backbone (gdb)",
            "type": "cppdbg",
            "request": "launch",
            "externalConsole": false,
            "miDebuggerServerAddress": "192.168.35.2:2345",
            "cwd": "${workspaceFolder}/Firmware",
            "preLaunchTask": "4-attach",
            "miDebuggerPath": "/opt/azurespheresdk/Sysroots/4+Beta2001/tools/sysroots/x86_64-pokysdk-linux/usr/bin/arm-poky-linux-musleabi/arm-poky-linux-musleabi-gdb",
            "program": "${workspaceRoot}/Firmware/out/debug/exe.unstripped/eGYM_Backbone.out",
            "MIMode": "gdb",
            "postDebugTask": "kill",
            "setupCommands": [
                {
                    "description": "Enable pretty-printing for gdb",
                    "text": "-enable-pretty-printing",
                    "ignoreFailures": true,
                }
            ]
        }
    ]
}
```

The problem here is that `4-attach` attach is where the telnet connects, and that session is alive blocking the debugger from move forward and attach to the client device on 2345. If I add `&` to the telnet call, is move forward, but any call that requires the `stdout` like `Log_Debug()` for example will crash.

So, is there a way to make this to work without requiring the telnet or if it is required, to have it run on VSCode in background without block the rest of the debugger session?

Thanks!

johnnyhli, 2020-03-06:
Hi @galvesribeiro - It seems that launch.json you shared is not  from AppSamples. Do you mind to try it again using the launch.json from AppSamples? Thanks.

johnnyhli, 2020-04-16:
Closed the issue, since there is no response for more than a month.

closed 2020-04-16 by johnnyhli