Azure/azure-sphere-samples #126: ADC value is always maximum

matsujirushi, 2020-02-29:
Maximum value is always displayed even if I change the potentiometer.
Code is [here](https://github.com/Azure/azure-sphere-samples/tree/master/Samples/ADC/ADC_HighLevelApp).

```
The out sample value is 2.490 V
The out sample value is 2.493 V
The out sample value is 2.495 V
The out sample value is 2.498 V
The out sample value is 2.500 V
The out sample value is 2.483 V
The out sample value is 2.485 V
The out sample value is 2.488 V
The out sample value is 2.490 V
The out sample value is 2.493 V
The out sample value is 2.495 V
```

AzureSphereTargetApiSet is 4+Beta2001 and SDK is 20.01.

johnnyhli, 2020-03-02:
Hi @matsujirushi – Can you share what hardware you were working on? If you are using MT3620 development kit, can you try a different ground pin? i.e. H1.2, H3.2, or H4.2 pin (please check [here ](https://docs.microsoft.com/en-us/azure-sphere/hardware/mt3620-user-guide#interface-headers)for more information). If you still have the issue, do you mind to share a diagram or a picture of your circuit?  If you want to share it privately, please sent to AzCommunity@microsoft.com with a link back to this issue. Thanks.

matsujirushi, 2020-03-03:
Hardware connection is same [here](https://github.com/Azure/azure-sphere-samples/blob/master/Samples/ADC/ADC_HighLevelApp/README.md#set-up-the-adc-connections).

[image]

When cutting the blue wire and H2.11, center of potentiometer voltage is 1.25V.
But when the blue wire and H2.11 are connected, center of potentiometer voltage is 3.3V.
Looks like H2.11 is not an analog input.

johnnyhli, 2020-03-04:
Hi @matsujirushi - I can't reproduce your case. Do you have another board that you can try?  Thanks.

matsujirushi, 2020-03-05:
Hi,
I got another board and tested.
It's fine.

Maybe first rdb broken...

Thanks for your help.

closed 2020-03-05 by matsujirushi