Azure/azure-sphere-samples #127: VS19 CMake errors

Raffa50, 2020-03-20:
Os: Windows 10 V10.0.18362 build 18362
Visual studio: 19 V16.5.0
Azure SDK for VS: 20.01

After creating a new project (example blink) or opening the sample, under the run menu (green arrow for running programs) I see no device
also I get a lot of CMake errors

```
Errore		CMake Error: Could not find cmake module file: C:/Users/Reloa/OneDrive/Reply/Azure Sphere - XCH20/AzSphere Reply Smart Valve V3/out/ARM-Debug-4/CMakeFiles/3.16.19112601-MSVC_2/CMakeReplyCompiler.cmake				
Errore		CMake Error: CMAKE_Valve_COMPILER not set, after EnableLanguage				
Errore		CMake Error: Could not find cmake module file: CMakeV3Information.cmake				
Errore		CMake Error: Could not find cmake module file: CMakeDetermineReplyCompiler.cmake				
Errore		CMake Error: CMAKE_Smart_COMPILER not set, after EnableLanguage				
Errore		CMake Error: Could not find cmake module file: CMakeReplyInformation.cmake				
Errore		CMake Error: Could not find cmake module file: C:/Users/Reloa/OneDrive/Reply/Azure Sphere - XCH20/AzSphere Reply Smart Valve V3/out/ARM-Debug-4/CMakeFiles/3.16.19112601-MSVC_2/CMakeValveCompiler.cmake				
Errore		CMake Error: CMAKE_V3_COMPILER not set, after EnableLanguage				
Errore		CMake Error: Could not find cmake module file: CMakeValveInformation.cmake				
Errore		CMake Error: Could not find cmake module file: CMakeDetermineValveCompiler.cmake				
Errore		CMake Error: CMAKE_Reply_COMPILER not set, after EnableLanguage				
Errore		CMake Error: Could not find cmake module file: C:/Users/Reloa/OneDrive/Reply/Azure Sphere - XCH20/AzSphere Reply Smart Valve V3/out/ARM-Debug-4/CMakeFiles/3.16.19112601-MSVC_2/CMakeV3Compiler.cmake				
Errore		CMake Error: Could not find cmake module file: CMakeSmartInformation.cmake				
Errore		CMake Error: Could not find cmake module file: CMakeDetermineV3Compiler.cmake				
Errore		CMake Error: Could not find cmake module file: C:/Users/Reloa/OneDrive/Reply/Azure Sphere - XCH20/AzSphere Reply Smart Valve V3/out/ARM-Debug-4/CMakeFiles/3.16.19112601-MSVC_2/CMakeSmartCompiler.cmake				
Errore		CMake Error: Could not find cmake module file: CMakeDetermineSmartCompiler.cmake				
Errore		CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_Smart_COMPILER		C:\Users\Reloa\OneDrive\Reply\Azure Sphere - XCH20\AzSphere Reply Smart Valve V3\Missing variable is		
Errore		CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_Valve_COMPILER		C:\Users\Reloa\OneDrive\Reply\Azure Sphere - XCH20\AzSphere Reply Smart Valve V3\Missing variable is		
Errore		CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_Reply_COMPILER		C:\Users\Reloa\OneDrive\Reply\Azure Sphere - XCH20\AzSphere Reply Smart Valve V3\Missing variable is		
Errore		CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_V3_COMPILER_ENV_VAR		C:\Users\Reloa\OneDrive\Reply\Azure Sphere - XCH20\AzSphere Reply Smart Valve V3\Missing variable is		
Errore		CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_V3_COMPILER		C:\Users\Reloa\OneDrive\Reply\Azure Sphere - XCH20\AzSphere Reply Smart Valve V3\Missing variable is		
Errore		CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_Smart_COMPILER_ENV_VAR		C:\Users\Reloa\OneDrive\Reply\Azure Sphere - XCH20\AzSphere Reply Smart Valve V3\Missing variable is		
Errore		CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_Valve_COMPILER_ENV_VAR		C:\Users\Reloa\OneDrive\Reply\Azure Sphere - XCH20\AzSphere Reply Smart Valve V3\Missing variable is		
Errore		CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_Reply_COMPILER_ENV_VAR		C:\Users\Reloa\OneDrive\Reply\Azure Sphere - XCH20\AzSphere Reply Smart Valve V3\Missing variable is		
Errore		CMake Error at C:\Users\Reloa\OneDrive\Reply\Azure Sphere - XCH20\AzSphere Reply Smart Valve V3\CMakeLists.txt:5 (project):
  No CMAKE_Reply_COMPILER could be found.

  Tell CMake where to find the compiler by setting the CMake cache entry
  CMAKE_Reply_COMPILER to the full path to the compiler, or to the compiler
  name if it is in the PATH.	AzSphere	C:\Users\Reloa\OneDrive\Reply\Azure Sphere - XCH20\AzSphere Reply Smart Valve V3\CMakeLists.txt	5	
Errore		CMake Error at C:\Users\Reloa\OneDrive\Reply\Azure Sphere - XCH20\AzSphere Reply Smart Valve V3\CMakeLists.txt:5 (project):
  No CMAKE_V3_COMPILER could be found.

  Tell CMake where to find the compiler by setting the CMake cache entry
  CMAKE_V3_COMPILER to the full path to the compiler, or to the compiler name
  if it is in the PATH.	AzSphere	C:\Users\Reloa\OneDrive\Reply\Azure Sphere - XCH20\AzSphere Reply Smart Valve V3\CMakeLists.txt	5	
Errore		CMake Error at C:\Users\Reloa\OneDrive\Reply\Azure Sphere - XCH20\AzSphere Reply Smart Valve V3\CMakeLists.txt:5 (project):
  No CMAKE_Valve_COMPILER could be found.

  Tell CMake where to find the compiler by setting the CMake cache entry
  CMAKE_Valve_COMPILER to the full path to the compiler, or to the compiler
  name if it is in the PATH.	AzSphere	C:\Users\Reloa\OneDrive\Reply\Azure Sphere - XCH20\AzSphere Reply Smart Valve V3\CMakeLists.txt	5	
Errore		CMake Error at C:\Users\Reloa\OneDrive\Reply\Azure Sphere - XCH20\AzSphere Reply Smart Valve V3\CMakeLists.txt:5 (project):
  No CMAKE_Smart_COMPILER could be found.

  Tell CMake where to find the compiler by setting the CMake cache entry
  CMAKE_Smart_COMPILER to the full path to the compiler, or to the compiler
  name if it is in the PATH.	AzSphere	C:\Users\Reloa\OneDrive\Reply\Azure Sphere - XCH20\AzSphere Reply Smart Valve V3\CMakeLists.txt	5	
```

PS: I had to set AzureSphereDefaultSDKDir because it was not setted by the installer, also there is something else missing?

johnnyhli, 2020-03-20:
Hi @Raffa50 - Happy to help and thanks for asking. We'll look into it.

gloveboxes, 2020-03-22:
Have you tried moving the project to a directory closer to root, for example c:/projects. CMake doesn't like long path names.

olegkhar, 2020-04-15:
@Raffa50 - Can you please share your CMakeLists.txt and CMakeSettings.json file so we can try to reproduce the issue?

Raffa50, 2020-04-15:
I have fixed by setting system variables:
AzureSphereDefaultSDKDir
AzureSphereTargetApiSet

closed 2020-04-15 by Raffa50