Azure/azure-sphere-samples #128: No data in IoT central device

piercalderan, 2020-03-24:
Everything seems to be OK. This is the log. The result from IoT Hub is 0 and nothing can be viewed in the temperature panel (always witing for data).
Hope you can help . Thank you.

Remote debugging from host 192.168.35.1, port 65357
IoT Hub/Central Application starting.
Setting Azure Scope ID 0ne000506D4
Opening SAMPLE_BUTTON_1 as input
Opening SAMPLE_BUTTON_2 as input
Opening SAMPLE_LED as output
[Azure IoT] Using HSM cert at /run/daa/df7ba4db-4a32-4bfd-b401-59f2846cc320
IoTHubDeviceClient_LL_CreateWithAzureSphereDeviceAuthProvisioning returned 'AZURE_SPHERE_PROV_RESULT_OK'.
Sending IoT Hub Message: { "Temperature": "30.30" }
INFO: IoTHubClient accepted the message for delivery
Sending IoT Hub Message: { "Temperature": "29.45" }
INFO: IoTHubClient accepted the message for delivery
IoT Hub Authenticated: IOTHUB_CLIENT_CONNECTION_OK
Sending IoT Hub Message: { "Temperature": "29.05" }
INFO: IoTHubClient accepted the message for delivery
Sending IoT Hub Message: { "Temperature": "30.00" }
INFO: IoTHubClient accepted the message for delivery
Sending IoT Hub Message: { "Temperature": "29.15" }
INFO: IoTHubClient accepted the message for delivery
Sending IoT Hub Message: { "Temperature": "29.90" }
INFO: IoTHubClient accepted the message for delivery
Sending IoT Hub Message: { "Temperature": "30.50" }
INFO: IoTHubClient accepted the message for delivery
INFO: Message received by IoT Hub. Result is: 0
INFO: Message received by IoT Hub. Result is: 0
INFO: Message received by IoT Hub. Result is: 0
INFO: Message received by IoT Hub. Result is: 0
INFO: Message received by IoT Hub. Result is: 0
INFO: Message received by IoT Hub. Result is: 0

bawilless, 2020-03-24:
If you see: "IoT Hub Authenticated: IOTHUB_CLIENT_CONNECTION_OK" then you're connected to the IoT Hub  in the IoT Central implementation.  Verify that the **case and spelling** for the telemetry key "Temperature" exactly matches the Temperature "field name" in your IoT Central "create telemetry" configuration.  This field is case sensitive.

piercalderan, 2020-03-24:
SOLVED,

Thank you.

Best,

johnnyhli, 2020-03-26:
Thanks for your question @piercalderan and thanks for you answer @bawilless. Glad to hear that the issue has been solved.

closed 2020-03-26 by johnnyhli