Azure/azure-sphere-samples #129: Memory Size on Azure Sphere MT3620

mehrdadhe, 2020-03-30:
I need to allocate a large memory(about 2.5MB) on AS using malloc() function which was not successful. I'm trying to understand the memory limitation on AS.
[This link](https://docs.microsoft.com/en-us/azure-sphere/app-development/application-memory-usage) mentions that the RAM size is 256KiB. In contrast the [MT3620 datasheet](https://d86o2zu8ugzlg.cloudfront.net/mediatek-craft/documents/mt3620/MT3620-Datasheet-v1.3.pdf) mentioned system memory of 4MB.
In my experiment, I was able to allocate a memory up to 700KB. So, what is the actual memory size for user application?
I couldn't find any other place to post this question and I appreciate if you could respond me.

JakeSays, 2020-04-11:
@mehrdadhe The official limitation is 256kb per application. It's interesting that you were able to allocate 700kb.

mehrdadhe, 2020-04-13:
@JakeSays you're right. The memory size is actually only 256KB. The 700KB is the size of my imagepackage which locates on the flash memory.

closed 2020-04-22 by johnnyhli