Azure/azure-sphere-samples #132: IntercoreComms RTApp does not build with VS Code

bawilless, 2020-04-27:
When I try to build and run the IntercoreComms_RTApp* example with VS Code I get an error when the project loads: "Missing property "program". [7,9]" on the launch.json file.  

I can correctly load, build and run the intercoreComms_HighLevelApp without any issues using VS Code so I'm pretty sure my system is configured correctly for VS Code builds.

* I'm using the VS 2019 SDK
* I don't see a "program" property in the *HighLevelApp*/launch.json file

Any idea how to correct this issue?

Thanks,
Brian

johnnyhli, 2020-04-30:
Hi @bawilless - Thanks for asking the question. We will look into it. Could you help me clarify the question? Did you have any issue on building other RTapp? Thanks.

bawilless, 2020-04-30:
I'm pretty sure that the intercoreComms example is the only MSFT Azure Sphere example that has a RTapp.  I can build and run this example if I'm using Visual Studio 2019, but not Visual Studio Code.  

Thanks for the help,
Brian

johnnyhli, 2020-04-30:
@bawilless There is another RTapp in [HelloWorld folder](https://github.com/Azure/azure-sphere-samples/tree/master/Samples/HelloWorld/HelloWorld_RTApp_MT3620_BareMetal). Can you try build that in VS Code? That would help us to better understand your issue. 

Thanks,
Johnny

bawilless, 2020-04-30:
Well now I can build and run both applications.  I'm not sure what changed, but it's working.  Thanks,

Brian

johnnyhli, 2020-04-30:
Hi @bawilless - Glad it works. I will close this issue for now. If you have similar issue again, please let us know. 
Thanks,
Johnny

closed 2020-04-30 by johnnyhli