Azure/azure-sphere-samples #135: Where to report issues with the sphere tooling?

JakeSays, 2020-05-15:
I have found a critical issue with azsphere_slattach. Where do I go to open an issue? I'm asking here because it's all I can find.

johnnyhli, 2020-05-21:
Hi @JakeSays - Thanks for reporting this issue. Are you using a Linux SDK? May I know which distro and which version you are using?  Thanks!

johnnyhli, 2020-06-29:
Hi @JakeSays - Do you still have the issue? An updated Azure Sphere SDK for Linux, version 20.04 Update 1, is now available. Please follow the [installation instructions ](https://docs.microsoft.com/en-us/azure-sphere/install/install-sdk-linux)to download and install this updated SDK.

JakeSays, 2020-06-29:
@johnnyhli I actually forgot I opened this. I had put my sphere work on the back burner, and am scheduled to get back to it sometime this week. At that point I'll check it out.

johnnyhli, 2020-07-22:
Hi @JakeSays - Do you still have the issue? If not I'll closed this issue. Thanks.

JakeSays, 2020-07-22:
@johnnyhli I haven't been able to reproduce it. I think it was related to upgrading my machine from ubuntu 19.10 to 20.04. It went away after I rebooted. It definitely was a bug but I doubt it'd ever occur under normal use.

johnnyhli, 2020-07-22:
Hi @JakeSays - Thanks for the help to reproduce the bug. It should be fixed in 20.04 Update 1 SDK. I'll close this issue for now. If it ever occurs or if you have any questions related, please feel free reopen this issue. Thanks.

closed 2020-07-22 by johnnyhli