Azure/azure-sphere-samples #137: Maximum filesize of resource files in image?

lokkitshmike, 2020-05-27:
I encountered an issue with resource files inside an app package.
I am adding a~750kb resource file by

`azsphere_target_add_image_package(${PROJECT_NAME} RESOURCE_FILES "file1.dat")`

in CMakeList.txt

With this setup, the application won't start when I start debugging from VS. VS starts sideloading the package and then fails with the message, that the package could not be sideloaded and that I should ensure, that development mode is enabled (which of course is enabled).

When I try to sideload the package manually with

`azsphere device sideload deploy --verbose`

it tells me, that the package was transferred successfully and then fails with the same message as in VS.

**The interesting thing about it is, that when I add two ~325kb files, it works perfectly**.

That's  uncool, because I am piggybacking an firmware image for an external MCU and I don't want to split up the firmware file into pieces manually before every build.

johnnyhli, 2020-05-29:
Hi @lokkitshmike - Thanks for bring this up. We'll look into it.

johnnyhli, 2020-06-02:
Hi @lokkitshmike - I was trying to repor your issue. I found that if I add two 375kb resource files, it still not working. Do you mind show more details about how you add two files? Thanks.

lokkitshmike, 2020-06-03:
I tried to reproduce the issue and I am now also not able to copy two of the 375kb files.
Maybe I was not 100% correct with the file sizes when I first opened the issue.

I am adding the files via CMakeList.txt with 

`azsphere_target_add_image_package(${PROJECT_NAME} RESOURCE_FILES "img_piggyback/test.txt" "img_piggyback/test2.txt")`

and Device update console is giving me

`Copying application to the device
Device would not sideload app: Installation failed; check if application development capability is enabled ('azsphere device enable-development') (DeviceClientException)`

The files are just 375kb of zeros in the 1st, and ones in the 2nd file.
What is the theoretical maximum amount of memory one can occupy for the read only storage?
Is there a way of reducing mutable storage space for expanding read only space?

johnnyhli, 2020-06-04:
Hi @lokkitshmike - I think this [link ](https://docs.microsoft.com/en-us/azure-sphere/app-development/application-memory-usage)could help.
> - 256 KiB RAM on the high-level core, reserved entirely for high-level application use. Up to 1 KiB of this space may be allocated for each shared buffer channel through which high-level applications and RTApps communicate.
> - 1 MiB read-only flash memory, which is shared between the high-level and real-time cores.
> - Read/write (mutable) storage, which persists when a device reboots. For information about mutable storage, see Using storage on Azure Sphere.

When debugging, the gdbserver plus your 750KB imagepackage goes over 1MB limit.

lokkitshmike, 2020-06-04:
Hi @johnnyhli - 

Ok thanks! So I think the issue can be closed.
What would you suggest for flashing an external MCU with an image >1mb? I am thinking about  just putting the md5 and url of the firmware in the image. I would then download the firmware to mutable storage, check its md5 and then flash it onto the external MCU using UART.

I could also think of streaming the download directly over the UART to the external MCU which would check the md5 by itself before flashing itself (OTA-capable ESP32).

@johnnyhli can you tell me the maximum usable amount of the mutable storage?

closed 2020-06-04 by lokkitshmike

johnnyhli, 2020-07-02:
Hi @lokkitshmike - Just update for your question. The mutable storage max is 64KB (see [here](https://docs.microsoft.com/en-us/azure-sphere/app-development/storage#application-manifest)), so downloading the firmware there won’t work.